iNavFlight/inav #10556: CRSF TX POWER not showing in OSD

dzaro-dev, 2024-12-24:
## Current Behavior

In INAV7, the CRSF TX POWER was shown correctly in the OSD in all types of MSP OSDs. Because I use DJI at the moment, I can only test if it works for the WTF.OS version and the O3 system, of which both show the same issue, no TX power value is shown, the field shows zero. 

Herer are some screenshots from an OSD recording:

| INAV7 | INAV8 |
|--------|--------|
| [image] | [image] |

## Steps to Reproduce
Flash INAV8 and enable CRSF TX POWER to be in the OSD.

## Expected behavior
Value should be the same as the one shown on the transmitter side (e.g. INAV telemetry widget).

## Suggested solution(s)
I can see some changes in the code related to this issue.

In INAV7, the `inav/src/main/io/osd.c` code was like this:

```
    case OSD_CRSF_TX_POWER:
        {
            if (!failsafeIsReceivingRxData())
                tfp_sprintf(buff, "%s%c", "    ", SYM_BLANK);
            else
                tfp_sprintf(buff, "%4d%c", rxLinkStatistics.uplinkTXPower, SYM_MW);
            break;
```

In INAV8, the `inav/src/main/io/osd.c` code was changed to this (I assume):

```
    case OSD_TX_POWER_UPLINK:
        {
            if (!failsafeIsReceivingRxData())
                tfp_sprintf(buff, "%s%c", "    ", SYM_MW);
            else
                tfp_sprintf(buff, "%4d%c", rxLinkStatistics.uplinkTXPower, SYM_MW);
            break;
        }

    case OSD_RX_POWER_DOWNLINK:
        {
            if (!failsafeIsReceivingRxData())
                tfp_sprintf(buff, "%s%c%c", "    ", SYM_MW, SYM_AH_DECORATION_DOWN);
            else
                tfp_sprintf(buff, "%4d%c%c", rxLinkStatistics.downlinkTXPower, SYM_MW, SYM_AH_DECORATION_DOWN);
            break;

```

I'm not sure if I'm looking at the right part of the code, but feel free to let me know if I'm wrong.

## Additional context
Here is a link to the CLI Dump:
[PasteBin](https://pastebin.com/FUWMj7Aq)

---

- FC Board name and vendor: MATEK F765
- INAV version string: INAV 8.0.0 RC3

MrD-RC, 2024-12-24:
That park of the code is basically the same. You don't need to worry about `OSD_RX_POWER_DOWNLINK` as that is for MSP RC only currently. `OSD_CRSF_TX_POWER` was renamed `OSD_TX_POWER_UPLINK`. The code within that block is basically the same. The only change is that is will continue to show the mW symbol if there is no power. So `SYM_BLANK` changed to `SYM_MW`.

It's possibly related to the setting of `rxLinkStatistics.uplinkTXPower`. Though that was not touched on the CRSF side in the MSP RC pull (https://github.com/iNavFlight/inav/pull/10451).

jaroszmm, 2024-12-24:
Yesterday I was flying my Swordfish on INAV 8 RC3 with O3 and had LQ and TX power as usual, same as on 7.1.2. DIFF all for update.

MrD-RC, 2024-12-24:
I'll check it out on my BARP. That has ELRS and Walksnail. Though I'll have to update that, which sucks. 

Ideally I should check it out on a proper Crossfire craft.

jaroszmm, 2024-12-24:
> I'll check it out on my BARP. That has ELRS and Walksnail. Though I'll have to update that, which sucks.
> 
> Ideally I should check it out on a proper Crossfire craft.

I am also sure 8 RC2 reports fine mLRS RX power on the OSD if that helps.

dzaro-dev, 2024-12-24:
Very strange, I've checked all settings yesterday and couldn't find why it is not showing. 

Could it be down to the ELRS version and how telemetry is being reported?

I was also trying out the ELRS Mavlink the other day and then switched back to normal ELRS CRSF.

Could that have messed things up? I know it messed up my telemetry sensors. 

I had to delete them from the radio and scan for sensors a few times.  

Each time the list of sensors was ordered differently - even the INAV telemetry widget was struggling.

dzaro-dev, 2024-12-24:
Additionally, why was the RSSI code changed to display `MAX `until take-off? Sorry, maybe slightly off-topic.

MrD-RC, 2024-12-24:
> Additionally, why was the RSSI code changed to display `MAX `until take-off? Sorry, maybe slightly off-topic.

It wasn't. It was changed to show `MAX` when there is 100%  RSSI. Before it would only go up to 99%. `MAX` was used instead of 100 to maintain spacing of OSD elements. As the RSSI % element only uses 2 digits.

glenn-venghaus, 2024-12-24:
fyi / extra datapoint for troubleshooting : 
running 8RC3 and DJI VISTA/WTFOS/TBS Crossfire and it shows tx power in OSD fine on the 3 wings i tested.
FC's matek f405-wmn (x2) and matek f405 wing v2.

dzaro-dev, 2024-12-24:
@glenn-venghaus That's because I'm using O3 Air Unit with Goggles 2. There is no icon for [mW] at the moment, so I have left it blank.

dzaro-dev, 2024-12-24:
@MrD-RC I've reflashed my ELRS receiver and trasmitter with 3.5.3 (currently the latest ELRS, but this is irrelevant because it worked fine in INAV 7.1.2), and the results were still the same. I've flashed back to 7.1.2 now because I can't get this to work. 

Could it be something with the DJI HD Compat 
Mode for O3 code that might be causing the issue?

MrD-RC, 2024-12-25:
No worries. So long as INAV is working correctly. I'll close this issue.

closed 2024-12-25 by MrD-RC

dzaro-dev, 2024-12-25:
Yes, absolutely, thank you for that. 

It still baffles me what actually happened. Must be some bad memory management or something.