iNavFlight/inav #10562: Demo Mode on inav 8.0.0 RC3 isn't working

rickgode, 2024-12-25:
## Current Behavior
inav 8.0.0 RC3 Demo Mode isn't working

## Steps to Reproduce

1. Start inav 8.0.0 RC3 configurator and then try to connect via Demo Mode
2.
3.
4.

## Expected behavior
Demo mode should work

Video: https://youtu.be/-6_Q0mUJH0A

dcan999, 2024-12-27:
Mine does the same, I figured it just wouldn't work until the official release, but who knows.

Jetrell, 2024-12-28:
A fix for this was added to the master a couple of weeks back.
You can download a nightly build and give it a try if you wish https://github.com/iNavFlight/inav-configurator-nightly/releases/tag/v20241216.86

rickgode, 2024-12-29:
That’s good to know, I appreciate all the effort put into inav, since I can’t develop, I feel the best way I can contribute is to test! Thanks again!

sensei-hacker, 2024-12-29:
> That’s good to know, I appreciate all the effort put into inav, since I can’t develop, I feel the best way I can contribute is to test! Thanks again!

Testing and documentation are super valuable!
All the features in the world don't do any good if nobody has written down how to use them, and besides they don't actually work right. :)

closed 2025-01-16 by b14ckyy