iNavFlight/inav #10565: Pitch oscillations in navigation mode INAV 8 RC3

Skuppersixseven, 2024-12-28:
## Current Behavior
Aircraft showing experiencing oscillations in Navigation mode  INAV 8 RC3

## Steps to Reproduce

1. Flew in manual - no issue
2. Flew in Acro - no issue
3. Flew in Angle - No issue
4. Flew in RTH - Pitch Oscillations 
5. Flew in Nav cruise -  Pitch Oscillations 

## Expected behavior
No pitch Oscillations 

## Suggested solution(s)
Find a way to remove Pitch Oscillations 

## Additional context
My CLI

diff all

# version
# INAV/SPEEDYBEEF405WING 8.0.0 Dec 13 2024 / 11:21:39 (253cf74b) dev
# GCC-13.2.1 20231009

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Timer overrides

# Outputs [servo]
servo 1 1000 2000 1494 -100
servo 2 1000 2000 1544 -100
servo 3 1000 2000 1524 -100
servo 4 1200 2000 1546 -100
servo 5 1000 2000 1460 -100

# safehome
safehome 0 1 301185944 -941779671

# Fixed Wing Approach
fwapproach 0 6000 950 0 -152 0 0
fwapproach 8 6000 500 1 332 152 0

# geozone

# geozone vertices

# features
feature SOFTSERIAL
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox GYRO_RAW
blackbox PEAKS_R
blackbox PEAKS_P
blackbox PEAKS_Y
blackbox SERVOS

# Receiver: Channel map

# Ports
serial 3 65536 115200 115200 0 115200
serial 4 33554432 115200 115200 0 115200
serial 30 1 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 1775 2100
aux 1 1 3 1300 1700
aux 2 12 3 1775 2100
aux 3 53 4 1300 1700
aux 4 11 4 1775 2100
aux 5 10 5 1775 2100
aux 6 28 2 1775 2100
aux 7 21 6 900 1175
aux 8 54 6 1800 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 6 valid
wp 0 1 301188519 -941782774 2000 0 0 0 0
wp 1 1 301208965 -941795395 5000 0 0 0 0
wp 2 1 301224861 -941757824 5000 0 0 0 0
wp 3 1 301166877 -941728594 5000 0 0 0 0
wp 4 1 301152303 -941760504 5000 0 0 0 0
wp 5 8 301183298 -941779639 0 0 0 0 165

# OSD [osd_layout]
osd_layout 0 0 49 0 V
osd_layout 0 1 12 0 H
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 7 24 1 V
osd_layout 0 9 12 11 V
osd_layout 0 11 2 18 V
osd_layout 0 12 6 18 V
osd_layout 0 13 36 10 H
osd_layout 0 14 2 0 V
osd_layout 0 15 13 10 V
osd_layout 0 22 25 5 V
osd_layout 0 23 17 3 V
osd_layout 0 24 24 6 V
osd_layout 0 25 34 8 V
osd_layout 0 27 35 11 V
osd_layout 0 28 11 18 V
osd_layout 0 30 12 16 V
osd_layout 0 31 7 0 V
osd_layout 0 37 12 17 V
osd_layout 0 38 1 17 V
osd_layout 0 44 0 0 V
osd_layout 0 46 39 1 V
osd_layout 0 48 0 19 H
osd_layout 0 53 7 17 V
osd_layout 0 85 36 10 V
osd_layout 0 96 36 10 H
osd_layout 0 99 18 7 V
osd_layout 0 108 33 3 V
osd_layout 0 110 48 1 V
osd_layout 0 112 48 2 V
osd_layout 0 129 35 0 V
osd_layout 0 140 33 4 V
osd_layout 0 144 21 17 V

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements
osd_custom_elements 0 1 0 0 0 0 0 0 0 "ORCA1"

# master
set gyro_zero_x = -10
set gyro_zero_y = -7
set gyro_zero_z = 12
set ins_gravity_cmss =  972.608
set acc_hardware = ICM42605
set acczero_x = -2
set acczero_y = 1
set acczero_z = 17
set accgain_x = 4093
set accgain_y = 4100
set accgain_z = 4097
set rangefinder_hardware = BENEWAKE
set align_mag = CW270FLIP
set mag_hardware = NONE
set align_mag_roll = 1800
set align_mag_yaw = 900
set baro_hardware = SPL06
set pitot_hardware = MS4525
set pitot_scale =  1.100
set blackbox_rate_denom = 32
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set failsafe_min_distance = 3048
set failsafe_min_distance_procedure = NONE
set failsafe_mission_delay = 6
set align_board_roll = 1800
set align_board_yaw = 1800
set bat_voltage_src = SAG_COMP
set small_angle = 180
set applied_defaults = 3
set gps_ublox_use_galileo = ON
set airmode_type = STICK_CENTER_ONCE
set nav_wp_load_on_boot = ON
set nav_wp_radius = 800
set nav_wp_enforce_altitude = 488
set nav_wp_max_safe_distance = 500
set nav_fw_wp_tracking_accuracy = 6
set nav_fw_wp_tracking_max_angle = 30
set nav_fw_wp_turn_smoothing = ON
set nav_land_maxalt_vspd = 100
set nav_land_slowdown_minalt = 305
set nav_land_slowdown_maxalt = 1841
set nav_emerg_landing_speed = 1349
set nav_min_rth_distance = 5000
set nav_rth_climb_ignore_emerg = ON
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 6096
set nav_rth_home_altitude = 6096
set nav_rth_use_linear_descent = ON
set nav_rth_trackback_mode = FS
set nav_fw_bank_angle = 45
set nav_fw_climb_angle = 30
set nav_fw_dive_angle = 30
set fw_loiter_direction = LEFT
set nav_fw_cruise_speed = 1565
set nav_fw_control_smoothness = 2
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set nav_fw_allow_manual_thr_increase = ON
set osd_video_system = BFHDCOMPAT
set osd_units = GA
set osd_time_alarm = 20
set osd_alt_alarm = 122
set osd_dist_alarm = 999
set osd_neg_alt_alarm = 0
set osd_current_alarm = 30
set osd_airspeed_alarm_min =  1207.000
set osd_airspeed_alarm_max =  3353.000
set osd_crosshairs_style = AIRCRAFT
set osd_camera_fov_h = 150
set osd_hud_radar_range_min = 9
set osd_switch_indicators_align_left = OFF
set name = ORCA-1
set nav_fw_land_approach_length = 30000
set nav_fw_land_final_approach_pitch2throttle_mod = 150
set nav_fw_land_glide_alt = 610
set nav_fw_land_flare_alt = 61
set nav_fw_land_glide_pitch = 1
set nav_fw_land_flare_pitch = 2

# control_profile
control_profile 1

set fw_d_pitch = 9
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 30
set nav_fw_alt_control_response = 40
set nav_fw_pos_xy_p = 55
set fw_level_pitch_trim =  3.000
set manual_rc_expo = 51
set manual_rc_yaw_expo = 30

# control_profile
control_profile 2

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# control_profile
control_profile 3

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 28
set motorstop_on_low = ON

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 50 0 -1
smix 1 2 0 50 0 -1
smix 2 3 1 50 0 -1
smix 3 3 2 -30 0 -1
smix 4 4 1 -50 0 -1
smix 5 4 2 -30 0 -1
smix 6 5 9 100 100 -1

# mixer_profile
mixer_profile 2

# Mixer: motor mixer

# Mixer: servo mixer

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 320
set battery_capacity = 5000
set battery_capacity_warning = 1000
set battery_capacity_critical = 500
set throttle_idle =  5.000
set nav_fw_min_thr = 1400
set nav_fw_max_thr = 1900

# battery_profile
battery_profile 2

set throttle_idle =  5.000

# battery_profile
battery_profile 3

set throttle_idle =  5.000

# restore original profile selection
control_profile 1
mixer_profile 1
battery_profile 1

# save configuration
save

# 

---

See above CLI

- FC Board name and vendor:
Speedybee F405 Wing

MrD-RC, 2024-12-28:
Duplicate of https://github.com/iNavFlight/inav/issues/10536

closed 2024-12-28 by MrD-RC

breadoven, 2024-12-29:
This is an ORCA model I take it using which control profile ?

Skuppersixseven, 2024-12-29:
No sir 
That’s only the name
It’s an atomrc swordfish