iNavFlight/inav #10566: Surface Mode is not working correctly (for many years)

utrinque96, 2024-12-29:
## Current Behavior
For years this issue was not solved. Surface Mode is not working correctly in INAV including last stable version. Range finders are useless in INAV. Endless discussions did not solve this issue.

## Steps to Reproduce
1. Install MicoAir MTF-02P or any other range finder e.g. MTF-01, MTF-01P. Turn on Surface Mode. It does not provide expected results. Many pilots reported dangerous abrupt altitude increase of their kwads after turning on Surface Mode.

3. I tested this bad behaviour of INAV in brand new 7 inch kwad, with new props, with zero crashes, zero noticable vibrations.

## Expected behavior
Range finder (called also "lidar", "sonar") is used ONLY in Surface Mode (MTF-02P at altitude 0-6 meters) in this mode other sensors like GPS and barometer should be ignored because range finder is much more precise. 

Surface Mode is to keep steady distance over uneven surface, not to keep constant altitude (GPS and baro are required for this).
At altitude below 6m Surface should have priority over Altitude Hold mode for safety reasons. It could help to avoid crash if kwad is headding towards slope of the hill.

## Suggested solution(s)
According to Paweł Spychalski INAV is also using accelerometer data which is causing many problems described by many pilots in different sources. I do not agree that vibrations are the reason of 95% of problems. Using accelerometer data for altitude assesment is.

Using accelerometer as source of altitude proven to be a bad idea. Remove it completely or provide option to turn it off and  ignore this data for altitude assessment.

Implement correct, working algorithm finally, please.

## Additional context

In this document Surface is listed correctly as: 
"Used to follow terrain, needs a rangefinder ( example sonar ). Not implemented properly and should not be used."
https://www.rcgroups.com/forums/showatt.php?attachmentid=11240541

Other reports and discussions in context to this issue (there are many more):
https://github.com/iNavFlight/inav/issues/10025
https://github.com/iNavFlight/inav/issues/6222
https://github.com/iNavFlight/inav/issues/606
https://github.com/iNavFlight/inav-configurator/issues/1210
https://github.com/iNavFlight/inav/pull/10314
https://github.com/iNavFlight/inav/pull/10314/files/79fa8f6e28dcd55d19f4905ac7269911ae1ac471
https://www.rcgroups.com/forums/showthread.php?3572803-Matek-Lidar-and-INAV
https://www.rcgroups.com/forums/showthread.php?2820715-INAV-punch-out-ALTHOLD

I have three sensors which can be used to keep steady altitude in Altitude or Position Hold Mode: GPS, baro, range finder. 
Why INAV is also considering accelerometer data after it proved so many times to generate problems and crashes for many pilots, for years?

Surface and AltHold in my case do not work.

Indoors GPS cannot be used for reliable Surface nor AltHold. There shuld be an easy option (e.g. stick command) to disable GPS and RTH fail safe and allow baro and sonar do its job in case of flying indoors. Currently it is not possible and requires disabling GPS in configurator or mobile app.

- FC Board name and vendor: DAKEFPVF405
- INAV version string: most recent official 7 and 8 release candidate

Is some developer in purpose disabling usage of range finders in INAV (due to the war in Ukraine?)? I am just curious.
Sabotage?

sensei-hacker, 2024-12-29:
OP seems to be confusing some things. Several unrelated things, actually.

The first linked issue is someone was flying three times as high as they had set for the maximum rangefinder range.  If you set the maximum rangefinder distance to one meter, and fly at 7 meters - it doesn't use the rangefinder.

The second is someone had an idea for a new mode, and then agreed such a thing isn't possible and it would be dangerous to try.

The third is eight years ago someone had trouble with position hold because their magnetometer to their FC - right next to the main power wires. Which has nothing whatsoever to do with surface mode.

Another linked PR is someone saying the accelerometer weight is *too low*.

So this "issue" is a list of links to random unrelated problems people had, some totally unrelated surface mode, most solved by assembling or configuring the aircraft properly.

@utrinque96  if you are having some specific problem, please describe YOUR problem and attach YOUR log. Linking to random issues that are totally unrelated to rangefinder or surface mode is not helpful.

The second link may be useful to you in that it explains why your "suggested solution" couldn't work.

utrinque96, 2024-12-29:
@sensei-hacker thank you for your answer.

Were you able to configure range finder for Surface Mode correctly? Do you fly with working range finder?

I am unable to configure MTF-02P to fly in Surface Mode indoors (without GPS) stable, keeping altitude consistently.

According to manufacturer: "INAV  7.1.1 may have some bugs that can not work properly with the sensor. You need to upgrade to version 7.1.2 which have been tested well."
https://micoair.com/optical_range_sensor_mtf-02p/

So there were problems till 7.1.1? Right?

My problem is that INAV algorythm uses accelerometer as source of altitude measurement when my model has two sensors measuring it already baro and sonar/lidar indoors and three outdoors (additionally GPS).

There are many reports from pilots having the same problem. Turning on this mode causes immediate strong gain or loss of altitude which is potentially dangerous and causes crashes.

Behaviour of model in Surface mode in INAV is bad, unpredictable, inconsistent, potentially dngerous in bigger models.

In my opinion this problem was not solved in7.1.1 nor in 7.1.2. 
Will it persist in 8.0 after release?

Definitely I am not an expert. I am not a developer. I just ask for checking INAV's algorithm for Surface modifier for other modes because many pilots have problems since at least 2021 when Paweł Spychalski published his video.

Honestly I was not aware that accelerometer is taken into account when assessing models altitude. Since Paweł confirmed it I am very surprised, slightly terrified when I discovered how algorithm is manipulating accelerometer readings before taking them for final calculations. 

Im my opinion

closed 2024-12-29 by utrinque96

sensei-hacker, 2024-12-29:
> I like your explanation of the difference between Surface and AltHold mode: [#10126 (comment)](https://github.com/iNavFlight/inav/discussions/10126#discussioncomment-10453585)

Thank you.

> Which of the two has higher priority? We cannot have both active at the same time. There seems to be a lot of confusion on this issue.

Right. As I mention in the documentation, you can't fly the purple line and the orange line at the same time. Surface mode altitude is orange line. Non-surface is the purple line.
You asked which takes priority - altitude hold or surface. Surface is a TYPE of altitude.  Setting surface by itself is like pulling up to a drive-in and ordering a "no onions". *What* without onions ? Maybe you want a *hamburger* without onion? A chicken sandwich without onion?  A chili dog? Similarly, you would order *altitude hold* with surface altitude.  It's not a matter of priority - "surface" by itself is meaningless.  You want "hold surface altitude" - it has to be both.

This may be the magic fix to your problem. If you were setting Surface without setting altitude hold, cruise or another altitude-controlled mode, just the adjective (modifier) "surface" wouldn't do anything. You would have just been ordering a "no onions".

On this topic, as also mentioned in the documentation, just as in English we put the adjective "surface" before the noun "altitude", the same is true in INAV. You need to select surface before selecting altitude (or simultaneously).  That is, you can't set altitude hold and then later say "I meant surface altitude".

> Is baro used in Surface Mode or not?

No.

> Were you able to configure range finder for Surface Mode correctly? Do you fly with working range finder?
> 
> I am unable to configure MTF-02P to fly in Surface Mode indoors (without GPS) stable, keeping altitude consistently.

I'm happy to set it up and send you the full config file, and photos of the physical mounting, working well, if you send me that sensor.  Note and-sh, in the issue you linked, discovered that the sonar sensors have a wide field of view, and will reflect off landing legs or whatever.

> According to manufacturer: "INAV 7.1.1 may have some bugs that can not work properly with the sensor. You need to upgrade to version 7.1.2 which have been tested well." https://micoair.com/optical_range_sensor_mtf-02p/
> 
> So there were problems till 7.1.1? Right?

The ONLY difference between 7.1.2 had to do with a setting related to the barometer. I happen to know this well because I'm the one who created 7.1.2, specifically to address the barometer issue. Since the baro isn't used in Surface altitude mode, it's not relevant to here.

> My problem is that INAV algorythm uses accelerometer as source of altitude measurement when my model has two sensors measuring it already baro and sonar/lidar indoors and three outdoors (additionally GPS).

The baro doesn't measure measure the distance to the nearest surface, perhaps the couch or countertop, so again it's not relevant in surface altitude hold.

> Another pretty recent discussion related to range finder did not solve the problem. Author gave up: #10487

And-sh discovered that their rangefinder gives reliable readings and it works well at 1-2 meters, not at 7 meters above the surface they were flying over. That's to be expected. As I recall (without double checking), the default values for both relevant settings is one meter.

You may have bought from a seller who advertised *up to* six meters. Note the *up to* advertised value is the best performance you'll ever get under ideal conditions. It's tested by pointing the rangefinder at a clean whiteboard in fairly dim indoor lighting. As the surface gets darker or less smooth, it will reflect less light and the performance will drop off accordingly. A smooth, medium gray metal surface will give range of about 3 meters.

utrinque96, 2024-12-30:
Thank you for clarification.

Definitely I am less confused now and have some hope to configure it to work finally.
Actually I did setup Surface modifier without Altitude Hold because someone pointed out that it has to be set before (or with) AltHold.

I am sorry for my earlier attitude. My only excuse is a bit of frustration. I hope my query and your answers will clarify this topic for other pilots having simillar issues. 

Happy New Year to awesome INAV Developers!

sensei-hacker, 2024-12-30:
I'm very glad to hear you are making good progress!
If you end up wanting help with tuning PID values, there is a tuning channel on the INAV Discord. People there, like Jetrell, know PID tuning far better than I.

I see you underlined the 6 meters at 90% reflectance. For reference, white office paper is 70% - 90%.  Derate as needed for the surfaces you intend to fly over.

Jetrell, 2024-12-30:
>Honestly I was not aware that accelerometer is taken into account when assessing models altitude.  I discovered how algorithm is manipulating accelerometer readings before taking them for final calculations.

>Using accelerometer as source of altitude proven to be a bad idea. Remove it completely or provide option to turn it off and ignore this data for altitude assessment.

@utrinque96 It may seem like the accelerometer isn't useful for XY and Z position control.. But in fact its an essential part of smoothing out vertical and horizontal position estimation.
Whether GNSS, Baro or even Lidar to a lesser degree. All these sensors are either limited by the best accuracy they can obtain. Or they are too slow to react to sudden position change.. Which is when a copter that is mechanically balanced and tuned, can take advantage of this.

>Im my opinion those sensors are unusable with current version of INAV and 8.0 and Surface mode is not working correctly now.

You are correct.. And this issue is due to the removal of `inav_w_xyz_acc_p` in 7.1.. With it being replaced with a dynamic clipping factor.. Which helps prevent _skyrocketing_ on poorly setup and imbalanced multicopters... But when higher transient vibration spikes do occur, it over clips the accelerometer weight. Causing the copter to wander off a bit in any direction, because the accelerometer weight becomes less significant. This is an example to you of how important the accelerometer is for position estimation.

But the above does need a fix. As does the way its effecting Lidar operation. As mentioned [here](https://github.com/iNavFlight/inav/pull/10314). So it hasn't gone without notice.. Its about who, when and how it will be fixed for best operation.

sensei-hacker, 2024-12-31:
@utrinque96  I thought you might find this interesting. This is one of the folks from PR #10314 , setting the accelerometer weight 1.0 (100%)

https://youtu.be/_PZ_zpTkVIU?si=B7iPVvkLkzh7W-lo

We have to see if we can get that adjusted so it doesn't change behavior when NOT using surface altitude, and get it merged for 8.1

utrinque96, 2024-12-31:
@sensei-hacker Thank you

utrinque96, 2024-12-31:
I seems default P and V values for Surface are very low:
[image]

sensei-hacker, 2025-01-01:
> I did increase both values to 90. 7inch quad seems to keep altitude indoors but after a while starts jojo movement up and down with increasing amplitude.

Going from 3.5 to 90 is a 2,400% increase. I would expect bad behavior when you increase ANY setting to 24X its default value. You might want to test 5, or 7. Going up slowly and watching for the sweet spot.

utrinque96, 2025-01-01:
> Going from 3.5 to 90 is a 2,400% increase. I would expect bad behavior when you increase ANY setting to 24X its default value. You might want to test 5, or 7. Going up slowly and watching for the [...]

I did change weight to 90/100 for both values to check if sonar can hold altitude above flat urface.
Those values are not PIDs.
I wonder what sensor prowides 10% missing weight?

[image]

and-sh, 2025-01-01:
Everything works, there are no noticeable errors in the code. Tuning is needed, an ideal barometer, and some average drone similar to the one on which it was debugged. not very powerful, not very weak, not very big, not very small, not very heavy, not very light.

utrinque96, 2025-01-01:
> Everything works, there are no noticeable errors in the code. Tuning is needed, an ideal barometer, and some average drone similar to the one on which it was debugged. not very powerful, not very weak, not very big, not very small, not very heavy, not very light.

sensei-hacker confirmed earlier:
[image]

and-sh, 2025-01-01:
The source code is open. It is not very complicated and is easy to view on git. All mathematics comes down to addition, subtraction and multiplication. If you want to understand what is happening, there is no other way.

and-sh, 2025-01-01:
You may start here
https://github.com/iNavFlight/inav/blob/8.0.0-RC3/src/main/navigation/navigation_multicopter.c#L77

utrinque96, 2025-01-01:
> You may start here https://github.com/iNavFlight/inav/blob/8.0.0-RC3/src/main/navigation/navigation_multicopter.c#L77

Interesting. Thank you. Last time I coded was Z-80 assembler in '80s.
Karnigan Ritchie C was my favourite book I never finished.
I got PhD in neuropharmacology in 2005, it was less complicated than coding.

and-sh is right it seems the only way to understand what is happening behind the curtain.

sensei-hacker, 2025-01-01:
navigation_pos_estimator_agl.c gives the altitude AGL (from the surface)

ttps://github.com/iNavFlight/inav/blob/68ee1bbf1a119873f141edd1e0bef3c1fdb72198/src/main/navigation/navigation_pos_estimator_agl.c#L71

utrinque96, 2025-01-01:
I have to go to basics, I just learned from the code about breaking mode.
https://www.youtube.com/watch?v=gaiTBFD1ZrA

and-sh, 2025-01-01:
You welcome. Think of fc not as a black box with magic that makes a piece of plastic fly, but as a set of simple interconnected algorithms that manipulate numbers. 
Wery usefull are debug modes.  For example 
https://github.com/iNavFlight/inav/tree/6d6f8b8b8a173add5addb3a771bf597aba744512/src/main
https://github.com/iNavFlight/inav/blob/68ee1bbf1a119873f141edd1e0bef3c1fdb72198/src/main/navigation/navigation_pos_estimator.c#L552
You can see what happens to the parameters inside the algorithms.

Basics is PID controllers.

utrinque96, 2025-01-02:
Off topic, but can be relevant.

I wonder if this technology is available to ordinary people?
Paweł Spychalski already mentioned to be approached quite often by fishy "companies" developing military equipment.

[image]

https://www.facebook.com/people/Kui-Wong/pfbid02KVp1dxi4mY6uRGt78foWM14QVAE7NwkswZFbF2qGgEp7N3PVmhrLhSLU8n6KUZal/?mibextid=LQQJ4d

https://www.youtube.com/@DaDa1313-n3d

They play with powerfull toys:
[image]

And discuss in Russian (we can discuss in Chinese using Google translator of course):
[image]

Actually it is possible to order from China all parts for 13 inch kwad for around $240-$250 including relatively cheap X4214 TM4214 380KV 6-8S motors.

sensei-hacker, 2025-01-02:
Yep, that's the thing with any new technology - electricity, gasoline engines, airplanes - people will use it for whatever they choose to use it for.

utrinque96, 2025-01-02:
It is far beyond my skill, but interesting finding. Seems Potensic Atom is wide open for reverse engineering:
[image]
https://www.reddit.com/r/Potensic/comments/1diwuxo/potensic_atom_se_teardown_and_discussion/?rdt=37534
I assume without sorce code it is very difficult to do. Anyway someone was trying.

utrinque96, 2025-01-06:
This project proves proximity sensor can work very well.
https://www.youtube.com/watch?v=dcu0jODIlWU

I get that drone can move backwards, sideways and implementation can be more complicated.

macflyhub, 2025-01-10:
@utrinque96 - If you get this working please let me know your settings etc.. Then hopefully I can finally fly with the MTF-01P. I have tried so many times to get this right. I am using a 5 inch quad.
@sensei-hacker - You take so much time to answer in detail. I am blown away. Amazing as always.

utrinque96, 2025-01-18:
> [@utrinque96](https://github.com/utrinque96) - If you get this working please let me know your settings etc.. Then hopefully I can finally fly with the MTF-01P. I have tried so many times to get [...]

Probably Ivwillvgive up on INAV and move all my kwads to Betaflight. Today I was flying 4 of my kwads. Gofilm 20, Potensic 3axis, 7inch and 10inch.

Small 2 inch kwad fly much better on Betaflight. I have 3 of them Gofilm 20 with Moonlight VTX&camera,2without Moonlight, one of them has small GPS receiver.

7 and 10 inch kwads fly wery well on INAV but I will probably limit sensors I use to gyro, accelerometer and GPS.

I was trying to use magnetometr, barometer, flow meter and sonar in 7 and 10 inch but INAV implementation and algorithms in my case were erratic, dangerous, not predictable and not reliable. 

Definitely I will play wit RTH, waypoints and modes utilising GPS. 

Because Betaflight has RTH I will focus on setting some of my kwads for freestyle, hard flying and bigger 10 and 13 inch for long range cruising, efficiency, low weight, low Amperage.

utrinque96, 2025-02-02:
I was curious and flashed INAV 8 on my 7 inch. It seems Surface modifier dissapeared. Am I right?

I wonder if config for Optical Flow & Rangefinder described by Oscar is still valid?:
https://oscarliang.com/setup-optical-flow-rangefinder-inav/

EZtune works fine. I tested only Angle mode indoors. I will not risk testing Althold and Poshold modes indoors. I have to wait for the opportunity to fly in the field to test them.

macflyhub, 2025-02-02:
@utrinque96 Funny you asked today.. I literally tried inav 8 with surface mode on (yes I had it on mine) and guess what.. still didn't work outside (didn't test inside). Also I used Oscar's guide in the very beginning.. but not sure if its accurate or even valid anymore as you said. I also asked MicoAir on Discord but no reply yet for a week or so but they may be on holiday in China.
Also I was pretty sure the quad (LR4) flew worse generally on inav - so switched back to latest BF and finished the setup too late to fly today but will have a go tomorrow if I have time after work.

utrinque96, 2025-02-02:
@macflyhub lets stay in touch. 
I was recently flying 2, 7 and 10 inch in the field. it was cold and I did not have enough time to connect to configurator and test different settings. I still have to limit throttle in 10 inch because in full throttle it draws over 80A and I have 60A ESC. For 7 inch 60A ESC seems to work fine. In 13 inch I plan to use 80A ESC and limit max throttle. 

I am looking for good guide setting ESC timings etc. to get best effciency for long range/ long flight times.

I swiched to 15 deg timing from higher timing in 2 inch after watching "ULTIMATE ESC Settings for Tiny Whoops!" by Chris Rosser. I did not notice lower curent draw nor longer flight times.
https://www.youtube.com/watch?v=EhYKeZfSQIw&t

My 2 inch is more cinewhoop based on Gofilm 20 frame than tiny whoop.

I use alternative parts:
https://www.rcgroups.com/forums/showpost.php?p=53210635&postcount=5612

OscarKator, 2025-10-29:
Well, 5 years after giving up drones due to lack of time (and excessive regulations), I build my 4" drone with Inav 8, and before intalling the gps, (and waiting for the hd wifi link), I try to install and work with the optical flow Mtf-02p (from Micoair), only to see similar problems that I had years ago. Carefully following the instructions for Inav from Micoair, Oscar Liang and another fans, and as soon I enable Surface Mode the drone fly away and crashes into the ceilling. 
The motors do not respond (if you are quick to try it).  Sad to see that there are stills problems. Always that I change to Betaflight, or Arducopter, I return to Inav, I like it, but something like a lidar and optical flow should be today as easy to install like a gps, There are plenty of cheap drones that use those sensors without any problems. 
Try seven times, with full erases and new installations of inav 8, perfect calibration  and fine data in debug, good orientation and wiring,  more than 2cm to floor begin fly,... testing differents flight/navigation modes,... seven crashes in the ceilling, before I change to "test mode"  attaching the drone to the table to observe the motors going crazy.

Board: flywoo aio 12A with Inav flywoo405pro target, , 2s lión, 4" propeller, elrs., 1306/3100kv motors

sensei-hacker, 2025-10-29:
> Well, 5 years after giving up drones due to lack of time (and excessive regulations), 

Can you please post a photo of your build?

OscarKator, 2025-10-30:
[image]
[image]