iNavFlight/inav #10569: Add support for JETI MULI6S cell voltage monitor

jaroszmm, 2025-01-02:
## Current Behavior
Up to now INAV has a decent cell voltage calculation based on autodetection of pack voltage on powering up
## Desired Behavior
Ad option (with external sensors) to monitor each pack cell separately and allow to either display all the separate cell voltages on OSD or at least make the lowest cell voltage appear on the OSD in the "cell voltage" slot

## Suggested Solution
Add option to read data from JETI Muli 6S EX external sensor which provides telemetry data to JETI radios with cell voltages.
I know INAV works with JETI system so I am pretty sure it is doable to read the data from that sensor somehow through UART probably.

https://www.jetimodel.com/katalog/muli-6s-module-ex-1.htm

## Who does this impact? Who is this for?
Anyone who wants to monitor closely his packs when pushing them to the limit. For me it would be most useful for LOS EDF jets where packs are stressed a lot and sometimes overall pack voltage might not suggest one cell going bad earlier etc.

RoadyFPV, 2025-01-05:
There are no devs who own jeti stuff as far as i know.
but why would you implement a muli, u can use the tx to display or alarm on critical cell voltage

jaroszmm, 2025-01-06:
> There are no devs who own jeti stuff as far as i know. but why would you implement a muli, u can use the tx to display or alarm on critical cell voltage

I can use the sensor with my JETI radio easily but if I use INAV for ease of launches of my EDFs (and extra stability) then I don't have a way to connect the sensor to the reciever directly. And since INAV works with JETI (I get INAV telemetry data on my JETI radio) I thought about implementing support for that sensor.

Which would benefit anyone as we could monitor every cell separately in case one goes bad in flight earlier, as cell voltage might not display it properly then. 

But I still need to check, maybe I can use both the MULI 6S and INAV with Jeti reciever at the same time.

RoadyFPV, 2025-01-06:
Wich RX do you use?

If u have a REX RX than u can use E1 and E2 for INAV (EXBUS) and the EXT Port for your sensor.

jaroszmm, 2025-01-06:
RSAT2 EX, using the EXBUS protocol as RC signal for the FC through an UART.

RoadyFPV, 2025-01-06:
ok, then you have no further possibility to connect a sensor.

jaroszmm, 2025-01-06:
So only walkaround now is a new reciever that can have more than one connection and connect it to INAV and separately add the MULI sensor.

RoadyFPV, 2025-01-06:
Yes, i think so.

I prefer the newer REX RX.
U have E1 and E2 with EXBUS and a EXT Port for Sensor

jaroszmm, 2025-01-06:
OK. I'll give it a try then.
But let's leave the feature request active. Maybe somebody will be able to add support for JETI sensors ;)

mmosca, 2025-01-17:
The likelyhood of this being added is very small. No one in the team has hardware to test/developer.

Is there at least documentation on the protocol anywhere?

RoadyFPV, 2025-01-17:
@mmosca Yes here are both. EXBUS and Telemetry protocol.

https://www.jetimodel.com/support/telemetry-protocol/

jaroszmm, 2025-01-19:
> The likelyhood of this being added is very small. No one in the team has hardware to test/developer.
> 
> Is there at least documentation on the protocol anywhere?

My limited knowledge was that since I fly some LOS planes with my JETI radio, JETI reciever and INAV on board and I can control my plane and also display a lot of INAV telemetry on my radio I thought the protocol and stuff is already known :)

jaroszmm, 2025-02-09:
> Yes, i think so.
> 
> I prefer the newer REX RX. U have E1 and E2 with EXBUS and a EXT Port for Sensor

I've realised I have one REX7 EX reciever. Just checked, I can connect it with EXBUS to my FC and also connect my vario to the Ext port.
Result - radio telemetry shows both INAV data and vario data. 
So I've already ordered the MULI 6s sensor, if this works, then I'll probably get myself one R3EX that will make a nice combo for INAVed LOS Jeti planes as I will have RC control via EXBUS with one wire and telemetry from the voltage cell sensor. All in one.