iNavFlight/inav #10572: Serial Gimbal settings via CLI won't accept negative values

jaroszmm, 2025-01-04:
****

## Current Behavior
Using CLI to center pan/tilt if using serial gimbal on UART (GM3 / C20T) gives very bad values.:

# set gimbal_pan_trim = 50
gimbal_pan_trim set to 50
# set gimbal_pan_trim = -50
gimbal_pan_trim set to 65486
# set gimbal_roll_trim = 50
gimbal_roll_trim set to 50
# set gimbal_roll_trim = -50
gimbal_roll_trim set to 65486
# set gimbal_sensitivity = -16
gimbal_sensitivity set to -16
# set gimbal_sensitivity = 15
gimbal_sensitivity set to 15

## Steps to Reproduce
Try setting negative value for gimbal trim for pan and roll.

## Expected behavior
Accept negative values if entered via CLI

---
# status
INAV/SPEEDYBEEF405WING 8.0.0 Dec 13 2024 / 11:21:39 (253cf74b) dev
GCC-13.2.1 20231009
System Uptime: 262 seconds
Current Time: 2041-06-28T03:04:00.000+02:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06, PITOT=VIRTUAL
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=UNAVAILABLE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1528
I2C Errors: 0, config size: 13455, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 11, cycle time: 507, PID rate: 1972, RX rate: 9, System rate: 9
Arming disabled flags: NAV HWFAIL RX CLI
  WAITING FOR GPS FIX
OSD: MSP DisplayPort [53 x 20]
VTX: not detected
GPS: HW Version: Unknown Proto: 0.00 Baud: 115200 (UBLOX Proto >= 15.0 required)
  SATS: 0
  HDOP:  99.990
  EPH :  99.990 m
  EPV :  99.990 m
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Max concurrent: 0

sensei-hacker, 2025-01-04:
Fixed in https://github.com/iNavFlight/inav/pull/10573

mmosca, 2025-01-10:
Fix has been merged.

closed 2025-01-10 by mmosca