iNavFlight/inav #10575: Autotrim

Mclags, 2025-01-05:
Small change suggested.
Disable continuous auto trim for auto launch mode.

I did something a little dumb that found this little snag. With continuous auto trim enabled, I had my craft sitting on the ground not level. I enabled launch mode but did not arm it. With a nav mode selected it appears that inav trimmed in an attempt to correct the roll axis. So when I armed and launched I had a severe roll trimmed into the craft that the launch mode did not level from.

Craft was sitting on the ground for approx 5 or 6mins in launch mode but unarmed.

Mclags, 2025-01-05:
> Rather than basing it on autolaunch. Maybe it should check to see if the aircraft is flying before trimming.

That would work too.

Carticiak, 2025-01-06:
Thanks for bringing this up. Same thing happened to me, I had to go back and manually set all my trims again, then send it up. Otherwise it was trying to bail hard to the left. 

Maybe it can use GPS data to confirm movement?

MrD-RC, 2025-01-06:
Not all aircraft have a GPS. There is a test within the firmware to estimate the aircraft is flying.

breadoven, 2025-01-06:
This shouldn't happen according to the code. Continuous Servo Autotrim is only active when Armed, it does nothing otherwise. Even if Armed it only trims when the plane is flying level in pitch and roll and also only if there's a valid GPS heading which means the plane needs to be flying with enough speed for the GPS to register a heading. So it's hard to see that the problem here was caused by Autotrim.

Mclags, 2025-01-07:
I am able to replicate this.
Plane on ground, in a nav mode, disarmed, push one wing down and hold it, can see the control surfaces slowly moving to counter the roll. Very slowly but they do move further and further from set mid point.

Mclags, 2025-01-07:
> I can't replicate it. I had launch mode enabled via switch, Continuous Servo Trim enabled, Poshold mode enabled and the plane was disarmed. There's no servo movement and the trims don't change.
>
> You'd need to provide a Diff to see if there's some setting causing this. Which Nav mode was enabled ?

Manual mode
Auto launch enabled
Still disarmed
Trims start moving.

I'll post my diff tomorrow afternoon.

b14ckyy, 2025-01-16:
@Mclags Any update on the diff? 

For documentation I link this bug here but that should have nothing to do with your issue. https://github.com/iNavFlight/inav/issues/9912

Mclags, 2025-01-23:
Sorry guys, life been getting in the way of the hobbies.
Diff below

# diff

# version
# INAV/MATEKF722WPX 7.0.0 Dec  5 2023 / 10:37:25 (895a4f31)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]
servo 2 1000 2000 1518 -100
servo 3 1000 2000 1518 -100
servo 4 1000 2000 1464 -100
servo 5 1000 2000 1536 100

# safehome

# features
feature -TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map
map RETA

# Ports
serial 2 33554432 115200 115200 0 115200
serial 3 2 115200 115200 0 115200
serial 4 4194304 115200 115200 0 115200
serial 5 0 115200 115200 0 115200

# LEDs
led 0 0,0::G:0
led 1 1,0::A:0
led 2 2,0::A:0
led 3 3,0::A:0
led 4 4,0::A:0
led 5 5,0::A:0
led 6 6,0::A:0
led 7 7,0::G:0

# LED color

# LED mode_color
mode_color 6 0 2
mode_color 6 1 6

# Modes [aux]
aux 0 0 0 1500 1900
aux 1 12 1 900 1100
aux 2 53 1 1500 1700
aux 3 11 1 1700 1900
aux 4 10 1 1900 2100
aux 5 21 1 1300 1500
aux 6 36 3 1725 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 48 4 H
osd_layout 0 1 1 13 H
osd_layout 0 2 0 0 V
osd_layout 0 6 47 17 V
osd_layout 0 7 24 16 V
osd_layout 0 9 2 9 V
osd_layout 0 11 3 16 V
osd_layout 0 12 1 11 H
osd_layout 0 13 3 10 V
osd_layout 0 14 47 5 V
osd_layout 0 15 46 9 V
osd_layout 0 22 26 2 V
osd_layout 0 23 3 2 V
osd_layout 0 28 24 10 H
osd_layout 0 30 12 14 V
osd_layout 0 31 47 6 V
osd_layout 0 32 2 17 V
osd_layout 0 35 2 14 V
osd_layout 0 38 2 15 V
osd_layout 0 40 3 1 V
osd_layout 0 41 15 0 V
osd_layout 0 42 36 0 V
osd_layout 0 62 10 11 H
osd_layout 0 63 10 10 H
osd_layout 0 97 21 0 V
osd_layout 0 110 47 4 V

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = -25
set gyro_zero_y = -20
set gyro_zero_z = -37
set ins_gravity_cmss =  339.400
set acc_hardware = MPU6000
set acczero_x = 96
set acczero_y = -42
set acczero_z = 1
set accgain_x = 4067
set accgain_y = 4097
set accgain_z = 4058
set align_mag = CW270FLIP
set mag_hardware = NONE
set align_mag_roll = 10
set align_mag_pitch = -1800
set align_mag_yaw = 2700
set baro_hardware = DPS310
set rssi_channel = 8
set rssi_min = 50
set rssi_max = 0
set serialrx_provider = CRSF
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set servo_protocol = SBUS_PWM
set small_angle = 180
set applied_defaults = 3
set gps_sbas_mode = SPAN
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 1500
set nav_wp_max_safe_distance = 500
set nav_rth_climb_first = ON_FW_SPIRAL
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_mode = MAX
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 25
set nav_fw_control_smoothness = 2
set nav_fw_launch_max_angle = 75
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_end_time = 5000
set nav_fw_launch_timeout = 30000
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set osd_video_system = AVATAR
set osd_alt_alarm = 1000
set osd_dist_alarm = 10000
set osd_hud_radar_disp = 4
set osd_hud_radar_range_min = 10
set i2c_speed = 800KHZ
set name = CLOUDS

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 28
set motorstop_on_low = ON

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 2 0 70 0 -1
smix 1 3 0 70 0 -1
smix 2 4 1 45 0 -1
smix 3 5 1 45 0 -1
smix 4 7 10 -100 0 -1
smix 5 8 15 100 0 -1
smix 6 4 2 45 0 -1
smix 7 5 2 -45 0 -1

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_ff_roll = 100
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 100
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 11
set pitch_rate = 9
set yaw_rate = 3

# battery_profile
battery_profile 1

set battery_capacity = 10000
set battery_capacity_warning = 4000
set battery_capacity_critical = 3000
set throttle_idle =  5.000
set nav_fw_cruise_thr = 1350
set nav_fw_launch_idle_thr = 1050

# save configuration
save

# end the command batch
batch end

ProfDrYoMan, 2025-04-17:
Any news on this?

Would love to keep servo auto trim enabled all the to ensure proper trim even if CG is moved or motor angle was adjusted.