iNavFlight/inav #10584: Request for new mode:  Airspeed selection

GTFreeFlyer, 2025-01-10:
## Current Behavior
User sets the throttle for cruise speed.

## Desired Behavior
A new, optional, mode that can be assigned to a channel, for example a 3-position switch, that allows the user to toggle between different cruise speeds.  Obviously, they wouldn't all be "cruise speeds", but the user could set one up for max. endurance speed, max range speeds, final approach speed, Vx, Vy, etc.  

## Suggested Solution
Mentioned above (and below)

## Who does this impact? Who is this for?
This would be mostly for FPV fixed wing pilots who want to toggle between different airspeeds during different portions of flight.  Personally, I fly FPV EDF jets and love to do long, stable final approaches.  It would be super neat to toggle the switch and have the FC hold my final approach speed for me until landing (auto throttle when throttle stick is anywhere above zero, then shuts off the airspeed hold when throttle stick is dropped to zero for the final flare.  Mode stays off until switch is cycled again, in order to prevent the mode from going active again when throttle is raised for taxi back to pit area).  Also, would love to set it for max range or max endurance depending on what I'm doing.  

## Additional context
iNav rocks with or without implementing this change!  Thanks for all the hard work!
But, if you want it to rock even more....  :)

b14ckyy, 2025-01-15:
This would need a full airspeed based throttle PID controller, that INAV does not have (yet). 
But you can change cruise throttle in flight if you enable `nav_fw_allow_manual_thr_increase`

you can then use logic conditions to create a throttle override based on a switch or based on Waypoint User-Actions to create change the cruise throttle value baysed on current mission waypoints.

Voltharion, 2025-01-22:
You can easily set a throttle override on a potmeter that only works when in cruise mode for example. It's what I do to set a fixed climbing angle in cruise or horizon mode.