iNavFlight/inav #10586: Matek F765 freezing up when arming

wilco1967, 2025-01-11:
## Current Behavior
Have a couple of planes, all iNav 8.0.0, various release candidates or nightly builds.
ONLY on my Matek F765 wing equiped MT, have some strange behaviour

All seems fine after I power up.
Servos respond to RC commands. 
Telemetry works. 
All fine
Wait for satelites.
(no mission loaded)

BUT, as soon as I arm, 4 out of 5 times, the FC freezes. 
Stops sending telemetry
Doesn't respond to movement, stick commands, nothing
OSD screen also becomes frozen.

Only way to get out of it is a power cycle.

Sometimes, lets say 1 in 5, it does arm normally, and I can fly without any problems.

Doesn't happen on my other planes / other FC

Flashed back to 7.1.2, and all works normal again....

---
# INAV/MATEKF765 8.0.0 Dec 13 2024 / 11:24:23 (253cf74b) dev
# GCC-13.2.1 20231009

# INAV/MATEKF765 8.0.0 Jan  1 2025 / 20:56:57 (34f7092e) dev
# GCC-13.2.1 20231009

Jetrell, 2025-01-11:
There were some changes made that improve F765 blackbox logging. 
I often use a Matek F765-WSE with the software after that commit. And I haven't experienced what you've reported.

So its likely an edge case bug, related to a setting or maybe hardware interaction.
Can you post your Diff All settings.

Did you have an SD card loaded in the FC ?
Was it logging flight data ? If yes, do you have a log ?

Another thing to try. Could be to remove the card and format it. Then load it back into the FC and let it write the partition.
Those FC couldn't do this correctly before.
Then check to see how it goes.

wilco1967, 2025-01-12:
Thanks for your respons.

I do have an SD card installed. 
[LOG00089.zip](https://github.com/user-attachments/files/18388747/LOG00089.zip)
attached are two BB logs. 89 likely was one of the failed arming attemps.
However, I'm not completely sure if this is the correct file. The date is 'strange'  to say the least. 
Not sure if it will even attempt to create a log if it freezes upon alarming, but it could also be, it attempts to start a log, and then fails somehow and freezes.

The 91 log was a flight were it armed just fine, and I made a flight.

Under firmware 7.1.2., I can't get MSC mode to work. In 8.0.0 MSC works fine
[diff.txt](https://github.com/user-attachments/files/18388760/diff.txt)

I tried to arm a couple of times, with stick override (no GPS lock), and that worked every time. (8.0.0RC3)

Then, after re-flashing to 8.0.0RC3, and restoring the diff, I could no-longer reproduce the problem.
I didn't take out or reformat the SD card yet.
it arms every time just fine.... 
Very strange.

Next, I've reformatted the SD card as you suggested. The FC has re-installed the 'freespace'  just fine.
But after formatiing, it froze again on the first arming attempt. however, subsequent attempts worked fine again.
here is the BB log files it created after re-formatting (not flying, just quick arming attempts)
[LOG00001.zip](https://github.com/user-attachments/files/18389007/LOG00001.zip)

If the problem re-occurs at the flying field, I'll try to take the SD card out.
For now, it failed only once on today's attempts. 
(today is not suitable flying weather, so it has to wait for another day)

I'll let you know if it reappears. 
Thanks again for the suggestions.

wilco1967, 2025-01-19:
Still not able to make a flight, due to weather / health.
But I noticed the SD card became corrupted while inserted in the FC
now on iNav 8.0.0RC4

MSC showed all kind of giberish files / directories
Also, inserted into the computer, all giberish

re-formated and card seems to be ok... 
H2TestW tested the card ok.

Jetrell, 2025-01-27:
@wilco1967 I'm not sure if you're still having issues.. I had a bit more of a look over your setting. And the only settings that stand out, that I don't use on my F765 are these :
`nav_fw_cruise_speed `
`cruise_power `
`idle_power` 
If its still a problem. Try removing these. Just to see if it can help narrow down a potential cause.

wilco1967, 2025-01-27:
Just made a flight yesterday, with zero issues. iNav 8.0.0 official.
However, I had left the SD card out, so if it is related to the SD card, it wouldn't show.

Those settings you mention, I had them set like that for a very long time. At least since iNav 6.0.0
But I can remove them, or set to default, no problem.

I will also re-install the SD card, see if the problem re-occcurs.
Don't know when the weather will cooperate for a new flight, but I can try a few times if it freezes upon arm on the bench.
I let you know if I find anything

wilco1967, 2025-02-01:
Today, made a flight with SD card installed, and settings as requested
(nav_fw_cruise_speed ,cruise_power , idle_power ,failsafe_mission_delay, all set to default / off / 0)

On first arming attempt, it immediate froze as before.

Did a power cycle, and on 2nd attempt, it armed just fine, and it completed a full flight without any issues. Still with the same settings, and SD card installed.

No BB log from the arming attempt which failed, but there is a BB log from the actual flight, which looks ok.

Few days ago, I did a number of arming attempts on the bench, and they all worked fine....

I'll be re-setting the settings as before, as it didn't seem to improve anything.

JWC1304, 2025-02-02:
is this not the same or variant of the same issue?

https://github.com/iNavFlight/inav/issues/10646

wilco1967, 2025-02-03:
> is this not the same or variant of the same issue?
> 
> [#10646](https://github.com/iNavFlight/inav/issues/10646)

Doesn't look like it.
There it apparently freezes after a random time.
In my case, it freezes exactly at the time of arming, and only sometimes.

sensei-hacker, 2026-02-25:
@wilco1967  stated that they could reproduce the issue "4 out of 5 times, the FC freezes."
If you're still around, to help narrow down the regression, here is a firmware build from the commit at the midpoint between 7.1.2 and 8.0.1:

**Commit:** `4e8bd5c6b36ad8cfb76da86ea719deef2855cca3`
**Date:** 2024-06-17
**Position:** ~815 commits after 7.1.2, ~814 commits before 8.0.1

**Download:** https://gist.github.com/sensei-hacker/1e6ec78488ff3ea763b51e86fe90f0f2
(Click "Raw" on the `inav_8.0.0_MATEKF765.hex` file to download)

---

@wilco1967 Could you flash this to your MatekF765 and test the arming freeze?

- **If the freeze IS present** → the regression was introduced in the first half (between 7.1.2 and this midpoint). We'll build from the first quarter next.
- **If the freeze is NOT present** → the regression was introduced in the second half (between this midpoint and 8.0.1). We'll build from the third quarter next.

wilco1967, 2026-02-28:
Did some testing, but the results are not very helpful I'm afraid.

The test firmware 8.0.0 I could not get the FC to freeze...
But I also could not get the FC to Freeze with current firmware 9.01.....

Tested about 10 times each version, SD card installed,  typically about 6..8 satelites when arming (as soon as would let me).
both re-arming a few times without power cycle, but also quite a few times with full power cycling....

Guess this doesn't prove anything....

The FC must have a 'good' day... We've seen that before.... I'll try a few times more tomorrow...

sensei-hacker, 2026-02-28:
Thanks so much for your efforts so far!
Maybe it a setting was changed in the time since the lock ups were happening.
I see there is a diff above from January 2025.

sensei-hacker, 2026-03-03:
I just fixed two bugs on F765.  It is POSSIBLE that the symptom discussed here may be related to either of these.
Hex files for testing can be found below.

https://github.com/iNavFlight/pr-test-builds/releases/tag/pr-11390
https://github.com/iNavFlight/pr-test-builds/releases/tag/pr-11391

I don't think either are very likely the cause of this issue, but either *could* be.
in particular this would probably NOT cover a failure before arming on a Matek.

sensei-hacker, 2026-03-03:
> Also, I re-formatted the SD card. After I re-inserted it, and powered up, it took about a minute where controls where very sluggish

I don't suppose you grabbed the log of the freeze first?