iNavFlight/inav #10587: How to properly connect and set up ESC if set in mixer to "Stabilised throttle" instead of directly motor output

jaroszmm, 2025-01-11:
I was wondering, if I am doing something wrong or is there something missing.

F405WMN with Delta mixer and one motor by default sets motor output to S1, S2 is empty (the same timer) S3 and S4 are for elevons.
I want to use S5 to control my ESC so what I do is I add in mixer tab another output as "stabilised throttle".

I need to keep the motor mixer on S1 because if I delete it then elevons are moved to S1/S2 and stabilised throttle goes on S3.

When I arm the plane and move my throttle stick the motor runs, so this is fine here.
However in the configurator tab where we can use configurator to run the motor with the slider I can't run my motor connected to S5.

Is this some kind of bug, maybe some "link" missing in programming? Or should I do something else. 
I do need the ESC to be on S5, basically as I want it to be on Dupont connectors and not soldered to S1 pads for ease of disconnection if needed.

[image]

sensei-hacker, 2025-01-11:
If you want your motor on S5, click the button to add a motor to S5.

If you click the button to add a *servo* on S5, then you are telling INAV that you put a servo on S5. As you discovered, the *motor* sliders control motors - not servos.

Just above the motor mixer, you'll see you can set different timers (output groups) to motor, servo, or to "auto" which puts the motors on the first timers.

jaroszmm, 2025-01-11:
[image]
This allowed to have control over motor on S5 as I wanted. But also moved my servos to S1 and S2. How can I now move servos to S3 and S4? Seems like INAV always pushes them to the first empty slot. I could move them by adding some empty mixers to S1 and S2 just to fill the outputs but not sure if this is the way (insert Mandalorian theme here)

jaroszmm, 2025-01-11:
Like that
[image]

sensei-hacker, 2025-01-11:
Yes, if you set the timers to "auto", it will take the first empty slot. If you want to skip a slot or two within those groups, adding a ghost servo or motor is fine.

jaroszmm, 2025-01-11:
Okay, so if I understand correctly - in my case there is no way to have Servo 1 and Servo 2 on S3 and S4 (Timer1) with S1 and S2 being empty. Something needs to be assigned to S1 and S2.

sensei-hacker, 2025-01-11:
> Okay, so if I understand correctly - in my case there is no way to have Servo 1 and Servo 2 on S3 and S4 (Timer1) with S1 and S2 being empty. Something needs to be assigned to S1 and S2.

Yeah, you'll assign something there - but that something doesn't have to actually exist. For purposes of clarity, I would probably set that up as a gimbal or similar.