iNavFlight/inav #10595: Inav can "disarm because of landing" even if the aircraft has not taken off.  (Inav 8.0.0 RC4)

MartinHugh, 2025-01-15:
## Current Behavior
Inav can "disarm because of landing" even if the aircraft has not taken off.

## Steps to Reproduce
1. Taxi out to the take off area
2. Reduce throttle to zero, to allow visual checks for people/dog walkers etc prior to take off
3. At this point the aircraft assumes a flight has completed and disarms the aircraft, despite not leaving the ground.

## Expected behavior
The aircraft should *not* disarm the aircraft for the reason of landing, if it has not yet taken off

## Suggested solution(s)
Do not disarm for reason of landing if the max altitude == 0

## Additional context
Settings for this feature are the defaults

# get nav_auto_disarm_delay 
nav_auto_disarm_delay = 1000
Allowed range: 100 - 10000

# get nav_disarm_on_landing
nav_disarm_on_landing = ON
Allowed values: OFF, ON

video of this happening
https://www.mhr.me.uk/disarmedbeforetakeoff.mp4

dump at :
https://pastebin.com/qdUGCEty

- FC Board name and vendor:
- INAV version string:
# version
# INAV/SPEEDYBEEF405WING 8.0.0 Jan 12 2025 / 19:18:44 (f0fcd155) 
# GCC-13.2.1 20231009

breadoven, 2025-01-15:
OK, looks like you met all the criteria for deciding it was flying without actually getting off the ground. In that case it will have checked for a landing once the throttle was low, triggering the landing disarm.

I guess a simple altitude check might work best to avoid this. Need to think about it and fix it for 8.0.

I'm surprised nobody else has reported this before. Guess most people hand launch.

b14ckyy, 2025-01-16:
lets pin that for 8.1
I don't feel good about changing existing functionality for a non-critical issue.

breadoven, 2025-01-17:
There's a very simple change to fix this problem, just looking at altitude gain of say 5m, However, this does highlight an issue when `inav_reset_altitude `is set to the default of `FIRST_ARM`. If you land and takeoff again from somewhere other than where you armed the reference altitude may no longer start at 0. This causes problems with the simplest fix for this issue and also for Launch, RTH etc if the new takeoff altitude is significantly higher than the arming altitude. Having said that if you are flying from somewhere hilly where this might happen you should be using `EVERY_ARM `for the above setting so in reality maybe it's not such a problem and the simple fix will work 99% of the time.

mmosca, 2025-01-22:
@MartinHugh can you test @breadoven 's PR?

MartinHugh, 2025-01-23:
> [@MartinHugh](https://github.com/MartinHugh) can you test [@breadoven](https://github.com/breadoven) 's PR?

Yes I will.  Weather is very bad today, but will try to get out, I dont necessarily need to take off to test this.

b14ckyy, 2025-01-23:
Sounds good. Take your time for a flight test. 
Just keep in mind that INAV will still disarm when you land and stop on the runway so you need to rearm for taxiing but when you have time to fly test more just send us an update. 
We will put this PR into 8.1 as long as no issue appears.

MartinHugh, 2025-01-25:
I managed a flight today, with lots of taxiing and stopping before take off.

Unable to reproduce the original problem.
Aircraft disarmed after landing and stopping, and indicated this in the post-flight stats.
No issues.

Thank you @breadoven

b14ckyy, 2025-01-25:
awesome. Will close here then :)

closed 2025-01-25 by b14ckyy