iNavFlight/inav #10606: Inconsistent arming with some BLHeli_32 ESC's due to 40 ms pause in DSHOT output caused by battery task

mluessi, 2025-01-17:
We have had reports of the ESC's not arming consistently with INAV 7.1.2. As far as I can tell, ithis only happens with ESC's that use AM32 microcontrollers and run BLHeli_32 32.9. Updating the ESC to 32.10 fixes the problem. However, updating is no longer possible as BLHeli has ceased operations.

I looked into this problem more and it seems to be caused by a 40 ms pause in the DSHOT output caused by the battery task (see `VBATT_STABLE_DELAY`). A hacky fix to the problem is to run the battery task once before enabling the DSHOT outputs. A proper fix would be to improve the battery task code such that the delay doesn't block other tasks. A screenshot of the outputs is below.

Note: Users have reported this problem with the BrainFPV version of INAV. However, I tested an OMNIBUS F4 flight controller with regular INAV and the 40 ms pause is present as well. So I assume the problem also affects some users that use regular INAV.

- FC Board name and vendor: BrainFPV RADIX 2, RADIX 2 HD
- INAV version string: 7.1.2-BRAIN-1

[image]

mluessi, 2025-01-17:
@mmosca I was able to reproduce it with a HobbyWing xrotor G2 65A 4-in-1 ESC, another user had the problem with a TMotor 65A 4-in-1 ESC. Getting rid of the 40 ms pause fixed the problem for both.

mmosca, 2025-01-17:
> [@mmosca](https://github.com/mmosca) I was able to reproduce it with a HobbyWing xrotor G2 65A 4-in-1 ESC, another user had the problem with a TMotor 65A 4-in-1 ESC. Getting rid of the 40 ms pause fixed the problem for both.

Yes, but what version of blheli? Does the pr fixes it for you?

mluessi, 2025-01-17:
@mmosca it happens only with blheli 32.9. I tested your fix with a HobbyWing ESC and it fixes the problem :)

closed 2025-01-22 by b14ckyy