iNavFlight/inav #10608: Receiver Tab doesn't work with the  RX Skystars SS24D ELRS 2.4G

skillcodeone, 2025-01-17:
## Current Behavior
With RX Skystars SS24D ELRS 2.4G, when go to Receiver tab doesn't work even if the is the correct communication with the RADIO working perfectly. The ERLS setting are the same on the betaflight firmware where there aren't problems.

Thanks

[image]

https://pastebin.com/dGn8q7v0

## Steps to Reproduce

1. Wire the RX Skystars SS24D to the FC  
2. Flash firmware inav 8.0
3. Set the serial on UART of RX
4. go to Receiver Tab

---

- FC Board name and vendor: GOKU GN 745 45A AIO 32bit (MPU6000 ) V3
- INAV/FLYWOOF745 8.0.0 Jan 12 2025 / 19:18:08 (f0fcd155) 
- GCC-13.2.1 20231009

mmosca, 2025-01-17:
Check your wiring. Configuration appears correct for rx in uart 2.

If it does not work, try a different uart and double check if the elrs rx is setup correctly to output crsf and not mavlink or another protocol.

skillcodeone, 2025-01-17:
> Check your wiring. Configuration appears correct for rx in uart 2.
> 
> If it does not work, try a different uart and double check if the elrs rx is setup correctly to output crsf and not mavlink or another protocol.

Hi, yes my RX on UART 2.
On RX side the protocol is CRSF

[image]

[image]

Flashing the Betaflight Firmware with same setting everything work perfectly...i don't understand....

mmosca, 2025-01-18:
Looks like you are using an unsupported INAV board, flashed with the wrong target.

Looking at Betaflight's targets, the AIO and the non AIO have separate targets.

[image]

skillcodeone, 2025-01-18:
> Looks like you are using an unsupported INAV board, flashed with the wrong target.
> 
> Looking at Betaflight's targets, the AIO and the non AIO have separate targets.
> 
> <img alt="Image" width="266" src="https://private-user-images.githubusercontent.com/23555060/404529376-678011d9-4fe5-46fc-85ba-12ac9653a5fe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzcyMjQ5MjEsIm5iZiI6MTczNzIyNDYyMSwicGF0aCI6Ii8yMzU1NTA2MC80MDQ1MjkzNzYtNjc4MDExZDktNGZlNS00NmZjLTg1YmEtMTJhYzk2NTNhNWZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE4VDE4MjM0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmOTg4ZjljM2FiNGZhZmFmZWNkYjJiYjBkZWQxMGUwMDQwYTFkOGYzYzAzYzQzYjUzYmViNjY0ZDc3YTE4MDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.HA7QeKnvMw3IC8OGWNBHPw2Ebrm_qetMcmLUfy5I1Uc">

I don't know if is full supported or no, i have flashed the version indicated on vendor's homepage ->  INAV Firmware: FLYWOOF745

[image]

skillcodeone, 2025-01-23:
Hi Unfortunately, i have activated for mistake also serial on UART1.
Leaving only the serial on UART2 everything works

Thank you so much

closed 2025-01-24 by MrD-RC