iNavFlight/inav #10613: Add ADC channels to programming framework

dcan999, 2025-01-18:
Not sure if it's even possible, but it would be nice to have access to the raw ADC channels in the programming operand tabs so we could use that voltage range input for analog sensors not supported natively, to show on OSD etc.

sensei-hacker, 2025-01-18:
This is something I've thought about doing. For a stall sensor or AoA sensor in my case. What would you be using it for?

dcan999, 2025-01-18:
The idea came from temp sensors, I am still working on getting my DS18B20 working, but the form factor could still be smaller, and there are a few things, like EGT on my turbine plane, that those can't support. There are copious amounts of converters that can convert many sensor outputs into analog voltages, including thermistors like a K-type that I could use for 800C EGT temps, but it wouldn't really be limited to that, it would just open up a large "DIY" section using the programming framework rather than waiting or asking for specific things to be implemented (like how native support for thermistors/analog temp sensors was declined)

I also designed a louver-type air inlet on one of my planes that operates in super low temps where I want to limit airflow into the fuselage to keep the battery from getting too cold, if I had the ADC (temp) input in the programming tab, I could map it to a servo output to control the louver and in turn the fuselage/battery temp.

DzikuVx, 2025-01-24:
I like this idea quite a lot!