iNavFlight/inav #10622: Add delay on MSP for disarm signal to allow digital systems to record stats screen

jaroszmm, 2025-01-19:
## Current Behavior
This issue is more noticeable in DJI but also a bit annoying in Walksnail.

Basically when landing and disarming digital systems stop recording. DJI instantly, Walksnail has a slight delay. As a result the DVR from DJI lacks stats screen and Walksnail has them for a second or so.

## Desired Behavior
After disarming INAV doesn't tell the VTX that it is disarmed for a short while while displaying stats first.

## Suggested Solution
I don't know if this is doable actually but my thinking was to somehow set a delay in MSP disarm message only for the UART that is connected to the digital VTX. So in result when we land and disarm INAV sends the disarm message everywhere, stopping motors, PID loop etc except for the one UART with digital system connected. This UART gets 5 second delay allowing INAV to properly display stats for a moment to be recorded before actually sending disarm signal to the VTX making it stop recording and going into low power mode.

## Who does this impact? Who is this for?
DJI and Walksnail users, not sure about HD Zero

## Additional context
There is a chance Walksnail devs implement the delay on their side (surely DJI won't haha) but maybe INAV can be faster?

b14ckyy, 2025-01-19:
I have talked about this with the other devs. 
INAV does not tell the FPV system anything. the FPV system requests status info from the FC with a MSP Message and the FC replies. In this case its the FlightMode Flags that are requested and contain the Arming state. 

So we have no way to delay the disarm information since the VTX gets the info directly from the FC status itself and not some arbitrary command.

MrD-RC, 2025-01-19:
More importantly, we shouldn't be delaying the arm status. Just to stop recording. It needs to report the correct state of arm as MSP is not just used for video systems. 

It is down to the video system to add an option to delay stopping recording after disarm.

closed 2025-01-19 by MrD-RC