iNavFlight/inav #10623: AIRSPEED DIST TO HOME AND RTH NOT WORKING

willyfpv, 2025-01-19:
Previuously everiting was working fine 

## Current Behavior
During a tuning flight I found no airspeed ,no distance from home and pos h

old or RTH not working ,I have taken off inside the radius of a Safehome 

## Steps to Reproduce
unable 

## Expected behavior
correct display of info on osd and correct functioning

## Additional context
Blackbox and DVR failed to load

Dump  here -->https://pastebin.com/44HiGDJb
---

- FC Board name and vendor: MATEK F405 wing v1
- INAV version string:
# version
# INAV/MATEKF405SE 7.1.2 May 31 2024 / 23:43:22 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

willyfpv, 2025-01-19:
I dont understand why I cant upload the files!

MrD-RC, 2025-01-19:
Works fine for me. This is likely a setup problem. So this should be a discussion, not an issue.