iNavFlight/inav #10629: Blackbox tab times out with Matek F405mini

rickgode, 2025-01-22:
Is anyone else having issues trying to access the blackbox in inav configurator 4.0 on a Matek f405 mini?  I have two, and both just time out.  I also have a Matek F405SE and a FlywooF411V2 and I can access the blackbox menu in configurator just fine.  Thoughts?

Ricky

mmosca, 2025-01-22:
Can you link the exact matek board you have?

It looks and behaves like a wrong a target situation.

mmosca, 2025-01-22:
Can you try #10630 ?

rickgode, 2025-01-22:
THANK YOU!!!  That looks to have done it!  This has been a bug since inav 4.0

[image]

rickgode, 2025-01-23:
Thanks @b14ckyy and @mmosca I hope the MatekF405 target gets merged

https://youtu.be/-r8S9HklKRQ

Ricky

mmosca, 2025-01-23:
Great! The fix will be in inav 8,1, once it is released :)

closed 2025-01-23 by mmosca