iNavFlight/inav #10646: MATEK H743 freeze after couple of minutes with INAV8

RoadyFPV, 2025-01-26:
I flashd INAV 8 to my MATEK H743
after a couple of minutes the FC frezze no Servo and configurator reaction, only a usb reconnect or disconnect the battery will help.
With INAV 7 no issues at all.

## Steps to Reproduce

1.Flash INAV 8
2.apply the modified diff / or set it up new
3.The FC freeze after a couple of minutes

## Expected behavior
It should not freeze like in INAV 7

## Suggested solution(s)
 Find with me a way to clear the issue, maybe u can give me instructions how to find it

## Additional context
[](https://pastebin.com/HxEp59E6)

---

# version
# INAV/MATEKH743 8.0.0 Jan 21 2025 / 16:13:24 (ec2106af) 
# GCC-13.2.1 20231009

- FC Board name and vendor: MATEK H743

Jetrell, 2025-01-26:
Can you please include a settings DIFF, not a DUMP

RoadyFPV, 2025-01-27:
Diff:  https://pastebin.com/Nm0Syd10 

It freeze with Battery only 
It freeze with USB only
It freeze totaly random. Sometimes after seconds, or after couple of minutes 
It freeze with insert sd card and disabled bblog
it freeze without sd card and disabled bblog

MrD-RC, 2025-01-27:
I see you are using Jeti, which you have reported elsewhere is locking up again. Don't you think that could also be the problem here?

RoadyFPV, 2025-01-27:
i dont know. 
INAV 7 is running without any freeze after i upgrade to an h743
now with inav 8 h743 freeze again
so i dont think its a JETI issue only

MrD-RC, 2025-01-27:
Have you tried setting serial RX to something like CRSF and physically removing the Jeti receiver?

RoadyFPV, 2025-01-27:
I will try it and set rx to sbus

MrD-RC, 2025-01-27:
So it seems like another Jeti related issue to me.

RoadyFPV, 2025-01-27:
can u compile me inav 8 with inav 7 compiler? lets see if this fix the issue like in inav 4

mmosca, 2025-01-28:
Try to test with the artifacts from https://github.com/iNavFlight/inav/pull/10653

No guarantees, so test in the bench first.

RoadyFPV, 2025-01-28:
Thanks, test is running...

What i also noticed some time ago, i have 16 channels but inav shows 24 channels on reciver tab.
if i have selected 24 channel mode in the tx then the green channel bars 17 to 24 are not moving and rock solid at 1500.
if i select non 24 channel mode, i think it was 2 bars, are jumping around.

RoadyFPV, 2025-01-28:
It freeze around 7 min

RoadyFPV, 2025-01-28:
Jea i think so..

i have a suspicion what it could be, but unfortunately i don't have the programming skills to implement it.

in the beginning with jeti, between rx and tx, i think it was a 2,4k resistor was needed. then marv-t in betafligh implemented it in the code so that no resistor is needed anymore.

maybe go back with the code so that the resistor is needed and test it?

mmosca, 2025-01-29:
I reached out to JETI to see if they are willing to provide a test system on loan, or if they have some sort of simulator we can use.

Otherwise, we may have to depend on someone in the community donating a radio + rx for testing. (It does not need to be the latest model, so if you anyone has an older JETI system that is compatible with EXBUS and is willing to donate to the INAV team, reach out!)

RoadyFPV, 2025-01-29:
Great, thank you.

I'm curious what answers now, don't think they'll go into it.

I might have a transmitter and rx available on loan, if jeti dosent respond.

May I contact you on discord?

mmosca, 2025-01-29:
> Great, thank you.
> 
> I'm curious what answers now, don't think they'll go into it.
> 
> I might have a transmitter and rx available on loan, if jeti dosent respond.
> 
> May I contact you on discord?

Sure, but I would wait for Jeti to reply.  I have no idea how long I would need the radio for, so I would rather know I am not preventing anyone from flying in the mean time :)

jaroszmm, 2025-01-30:
Meanwhile as I also have some JETI planes with INAV (using Jeti's exbus protocol) I could test out Inav 8 on some FCs. 

So what do I exactly need to do to cross check with you guys?

RoadyFPV, 2025-01-30:
I have right now flashd inav 8 on ma matekf722 wing.
and connected only rx to the fc.
i have the settings set without a cli restore.

lets see what happens.

@jaroszmm 
Flash INAV8
Use the uart tx pin and make sure telemetry is enabled
i use 16 channels in transmitter
and lets see if it freeze
the freeze is totaly random after some time. sometimes minutes or seconds or 20min...

maybe its a combination of jeti protocol an the other connected hardware

RoadyFPV, 2025-01-30:
@jaroszmm 
Pls check also 100Hz transmitter rate

RoadyFPV, 2025-01-30:
ok, freeze too with f722 with only rx connected  after around 20 min

JWC1304, 2025-02-01:
Hello,

I have the exact same issue with h743 and jeti on fixed wing, exactly the same as we had before somewhere in 6 or version 7.
With version 7.1 the system is very stable. Everytime again i am so happy with the new inav versioin and i always forget to look up the issues. Can this please be adressed, for i love the new feautures of inav 8

RoadyFPV, 2025-02-01:
@JWC1304 Can you pls try to disable the telemetry option in INAV configurator, and look if it freeze?

JWC1304, 2025-02-01:
> Meanwhile as I also have some JETI planes with INAV (using Jeti's exbus protocol) I could test out Inav 8 on some FCs.
> 
> So what do I exactly need to do to cross check with you guys?

if you use a matek controller and inav8 it will freeze after a few minutes of flying. the last time it was something with exbus jeti with made it freeze, basiccally losing signal.

JWC1304, 2025-02-01:
> [@JWC1304](https://github.com/JWC1304) Can you pls try to disable the telemetry option in INAV configurator and look if it freeze?

i would like toi but than i will have to sacrifice another plane or bird, i can only repeat it while actually flying.

RoadyFPV, 2025-02-01:
> > [@JWC1304](https://github.com/JWC1304) Can you pls try to disable the telemetry option in INAV configurator and look if it freeze?
> 
> i would like toi but than i will have to sacrifice another plane or bird, i can only repeat it while actually flying. Or can you tell me how to test this on the bench

No, it freeze on bench too with usb. pls confirm first the freeze with telemetry enabled and usb.
and then try disable telemetry with configurator

JWC1304, 2025-02-01:
> No, it freeze on bench too with usb. pls confirm first the freeze with telemetry enabled and usb. and then try disable telemetry with configurator

okay give me an hour (dinner in between )and i will report back asap

MrD-RC, 2025-02-01:
It's probably worth adding Jeti causes freezes in the known issues of the release notes.

RoadyFPV, 2025-02-01:
@MrD-RC Sounds reasonable, protects other user from lost her plane.

My test without telemetry runs for 2h15min now without a freeze and still running

RoadyFPV, 2025-02-01:
I think it could be a problem with the one wire communication misstiming or somewhat.

maybe we can try the old jeti exbus and telemetry code from the beginning.
there is a rersistor between rx and tx and the reciever is connected to rx.

https://github.com/betaflight/betaflight/blob/v3.3.3/src/main/rx/jetiexbus.c
https://github.com/betaflight/betaflight/blob/v3.3.3/src/main/telemetry/jetiexbus.c

after that, the code was changed so there is no resistor needed and connect directly to tx pin.

RoadyFPV, 2025-02-01:
3h and no freeze

JWC1304, 2025-02-01:
just to be sure mine is running now without ceasing and telemetry off in the configuration tab right? Will also tru to reproduce freeze with telemetry on, so far no freeze. I also assume it shpould be armed as well?! yes

RoadyFPV, 2025-02-01:
try first to reproduce the freeze with telemetry. arming is not required.
then disable telemetry and try again

JWC1304, 2025-02-01:
> try first to reproduce the freeze with telemetry. arming is not required. then disable telemetry and try again

ok WILCO

RoadyFPV, 2025-02-01:
I think we are on the right way to find the issue on the telemetry side.
But without Hardware on dev side it will be almost impossible.

JWC1304, 2025-02-01:
are you sure not arming? , because on the bench now for 6 minutes with telemetry and not freezing up, while in flight it is before the 5 minutes mark. ( i do not have gps lock, makes that a diffrencer because  i am inside not near a window)

RoadyFPV, 2025-02-01:
freeze is totaly random sometimes 20 min or 4min or 1h10min

RoadyFPV, 2025-02-01:
My setup for testing is matek f722 wing with only rx connected

JWC1304, 2025-02-01:
> My setup for testing is matek f722 wing with only rx connected

testing with matek h743, jeti transmitter, and trying to get gps now and arm and kmeep testing till it freezes.
I flew twice today with this setup, and it froze both times withing 5 minutes. Second time i was expecting it, firts time i was lucky :-)

JWC1304, 2025-02-01:
Yep it froze !! After 10 minutes, Call me nuts, but with me it is every time i am playing with throttle while armed ,it freezes, loses signal, etc. wil try to repeat this now without telemetry.

JWC1304, 2025-02-01:
Okay so far 10 minutes in without telemetry and gps and so far playing with throttle like a fool, but all systems seem to keep working and stable.

RoadyFPV, 2025-02-02:
@jaroszmm have you already been able to carry out a few tests

@JWC1304 can you pls try another one without telemetry and let it run for 1h or 2h?
And then again with telemetry? to be sure the telemetry part is the issue?

JWC1304, 2025-02-02:
> [@jaroszmm](https://github.com/jaroszmm) have you already been able to carry out a few tests
> 
> [@JWC1304](https://github.com/JWC1304) can you pls try another one without telemetry and let it run for 1h or 2h? And then again with telemetry? to be sure the telemetry part is the issue?

roger will do it again withpout telemetry for as long the battery will hold

jaroszmm, 2025-02-02:
> [@jaroszmm](https://github.com/jaroszmm) have you already been able to carry out a few tests
> 
> [@JWC1304](https://github.com/JWC1304) can you pls try another one without telemetry and let it run for 1h or 2h? And then again with telemetry? to be sure the telemetry part is the issue?

Sorry, not yet. Been busy with other stuff but might find some time today evening.

JWC1304, 2025-02-02:
Well i can confirm again without Telemetry i can not induce a freeze, within the 40 minute mark, i stopped than. Tried to induce it again with telemetry, on usb, without usb connected, disarmed, armed, with low throttle, etc. It seems it freezes up only when i give max throttle and than lower it again and than full again, thoise are the moments it freezes within minutes.(with telemetry) Without telemetry i can not make it freeze at all. (gps ios still connected btw)

RoadyFPV, 2025-02-02:
Yes, looks like the same issue wich comes out with the channge from gcc9 to gcc10.
But now we can confirm it has something to do with the telemetry part.

JWC1304, 2025-02-02:
> Yes, looks like the same issue wich comes out with the channge from gcc9 to gcc10. But now we can confirm it has something to do with the telemetry part.
> 
> This Issue is a real ghost. It comes and goes.....

for me there is definitely a correlation when armed and playing with my throttle over 50%, i do not get it,but that is when i can induce a freeze in minutes, whille sitting on the bench,armed and at low constant throttle it seems stable, very strange (needless to say this is with tlelemetry).

RoadyFPV, 2025-02-02:
another 3h test run without telemetry and no freeze.

yes, i tried out wiggeling the throttle stick with telemetry enabled and it freeze much faster. you are right.

I think that the different packages collide with each other (telemetry request and the channel data)

so that it leads to freezing when switching the rx/tx functions. 

all just a guess, i can't confirm it.
if i knew if i could somehow measure it with the oszi, i could try it

robertbulko, 2025-02-02:
Hi everyone, this problem seems to be exactly the same like when version 6 was released and new version of compiler was used. MATEK H743 was working properly only when telemetry was off or when INAV was compiled with version 9.3.1. I still use INAV 6 compiled with 9.3.1. and it work without any problems till now. I was very critical to developers at that time and was banned for my hot comments. History repeats itself but this time one of the devs is willing to take Jeti HW into his hands and try to correct the issue. I am glad to see such a progress and hope for Jeti users to see a light at the end of the tunnel.

RoadyFPV, 2025-02-05:
@robertbulko @JWC1304  can you guys pls test this file with telemetry pls? Only for Benchtest

[INAV 8 benchtest.zip](https://github.com/user-attachments/files/18676776/INAV.8.benchtest.zip)

robertbulko, 2025-02-05:
I am sorry, I won't do it. I have no spare H743 board for testing and do not want to disassemble the one i have in the plane. I'm happy it is working like it is.

JWC1304, 2025-02-06:
will try later today !

JWC1304, 2025-02-06:
> I am sorry, I won't do it. I have no spare H743 board for testing and do not want to disassemble the one I have in the plane. I'm happy it is working. (I mean using version 6.0 compiled on v9.3.1)

you do not need to offer it up just reverse to settings after testing

robertbulko, 2025-02-06:
> [@robertbulko](https://github.com/robertbulko) > I am sorry, I won't do it. I have no spare H743 board for testing and do not want to disassemble the one I have in the plane. I'm happy it is working. (I mean using version 6.0 compiled on v9.3.1)
> 
> you do not need to offer it up just reverse to your old settings and or version after testing

I did it before and I consider it useless unless some of devs really has JETI HW for testing. In my opinion, this kind of testing proves really nothing relevant.

robertbulko, 2025-02-06:
> > > [@robertbulko](https://github.com/robertbulko) > I am sorry, I won't do it. I have no spare H743 board for testing and do not want to disassemble the one I have in the plane. I'm happy it is working. (I mean using version 6.0 compiled on v9.3.1)
> > > you do not need to offer it up just reverse to your old settings and or version after testing
> > 
> > 
> > I did it before and I consider it useless unless some of devs really has JETI HW for testing. In my opinion, this kind of testing proves really nothing relevant.
> 
> That type of you dont realy help. Its better u will banned again.

Whatever, I am not willing to do the same thing and wait for different result. It is not very smart. As I said, the only relevant in my opinion is developer having the JETI HW. Until that time it is useless.

RoadyFPV, 2025-02-06:
2 test done with modified code.

1h30min no freeze with telemetry enabled
3h15min no freeze with telemetry enabled

RoadyFPV, 2025-02-07:
Next 3h50min run with no freeze
i think i have found something that reset the frames when its out of sync/time?

can a dev pls look over the code that i have edit? @MrD-RC or @mmosca

mmosca, 2025-02-08:
> Next 3h50min run with no freeze i think i have found something that reset the frames when its out of sync/time?
> 
> can a dev pls look over the code that i have edit? [@MrD-RC](https://github.com/MrD-RC) or [@mmosca](https://github.com/mmosca)
> 
> https://github.com/iNavFlight/inav/blob/master/src/main/telemetry/jetiexbus.c line 508 from 0 to 1 line 518 from 3000 to 1000

Can you submit it as a pull request? 
It also has the advantage of building things so more people can test.

RoadyFPV, 2025-02-08:
Unfortunately, I haven't done this yet and don't know exactly how it works.
Thats why i have attached the file.

what do you think does the value in line 518 do?
am i right in assuming that it has something to do with the fram reset and is now often reset?

i have build right now a new file with line back to 0 wich is for debug and i dont know where i can see it.

i will test it and see if it freeze

Did you get an answer from Jeti support?

mmosca, 2025-02-08:
No reply from jeti.

The easy way to create a pull request is to click on edit file on github, do you changes, and github will guide you through the process of creating the pull request.

RoadyFPV, 2025-02-08:
too bad, that's what I thought. jeti is not intrest on such things. 
Maybe u can contact elefun maybe they have a demo loan device for us?

i dont know is it the right way to pull something that is not proofed by a dev that understand the code?
if a dev can confirm that this change makes sense i can try to pull request this.

mmosca, 2025-02-08:
The pull request is the best way to have a dev look at it, and has the right tools to review the code.

[87 more replies]

closed 2025-03-26 by mmosca