iNavFlight/inav #10648: DSHOT sending 0048 on forward after reverse

pitts-mo, 2025-01-26:
## Current Behavior
INAV rover is sending a dshot signal 0048 burst when RC input transitions from neutral to forward after being in reverse.

Note: This does not happen on initial transition to forward from neutral or on attempts to transition to forward from neutral when the prior non neutral command was forward.

## Steps to Reproduce
 count, DShot Data, action 
   6561,  0000, **1** -Armed neutral
       76,  1048, **2** -transition forward
         3,  1050, increase forward
         3,  1054, increase forward
~
         3,  1056, decrease forward
       22,  1048, decrease forward
       54,  0000, **3** -transition neutral
         3,  0107, **4** -transition reverse
         3,  0114, increase reverse
~
         3,  0134, decrease reverse
         3,  0116, decrease reverse
11362,  0000, **5** -transition neutral
       41,  **0048**, **6** -transition forward _-this value should not be sent_
         3,  1052, increase forward
         3,  1061, increase forward

## Expected behavior
The first non zero command when transitioning from neutral to forward should be >=1048.

## Suggested solution(s)
mixer bug?

## Additional context
[INAV_8.0.0_cli_MAXX_20250126_084307.txt](https://github.com/user-attachments/files/18550669/INAV_8.0.0_cli_MAXX_20250126_084307.txt)

---

# INAV/MATEKF722MINI 8.0.0 Jan 21 2025 / 16:18:25 (ec2106af)

pitts-mo, 2025-01-31:
I have experimented in mixer.c. 
The most interesting things I found were strategically setting reversibleMotorsThrottleState = MOTOR_DIRECTION_FORWARD,  outputScaleMin = DSHOT_3D_DEADBAND_HIGH, and attempting to revert the dshot reverse fix. However, I have not established the understanding to make this work properly in both forward and reverse. 

dshot range values:
https://github.com/iNavFlight/inav/blob/03efa418bd7b64a8a1a190ccbe4b28af66decac4/src/main/flight/mixer.h#L30

dshot mix request:
https://github.com/iNavFlight/inav/blob/03efa418bd7b64a8a1a190ccbe4b28af66decac4/src/main/flight/mixer.c#L268

dshot direction/speed handling:
https://github.com/iNavFlight/inav/blob/03efa418bd7b64a8a1a190ccbe4b28af66decac4/src/main/flight/mixer.c#L377

specific fix for dshot reverse:
https://github.com/iNavFlight/inav/blob/03efa418bd7b64a8a1a190ccbe4b28af66decac4/src/main/flight/mixer.c#L583

pitts-mo, 2025-02-09:
Update:
Closing PR #10663 as it only moved the problem.

It feels like the detection of commanding forward vs reverse is somehow isolated from the dshot command output scaling in such a way as to allow new commands immediately after a change of direction to return results scaled for the old/prior direction. I have not identified the root cause but do question if this is a simple issue of refresh timing or a result of filtering?

Below is a sample of dshot command counts using PR #10663 around zero crossing as I slowly transition throttle between full forward and full reverse. This data set was started well after arming but for some reason I also see a burst of dshot command 0024 was sent after the first transition to forward had already begun.

[inav_pr10663 -crossing zero dhsot command counts.txt](https://github.com/user-attachments/files/18724721/inav_pr10663.-crossing.zero.dhsot.command.counts.txt)

sensei-hacker, 2026-02-01:
Before investigating this as a bug, I need clarification on expected behavior.

Looking at the INAV code (mixer.c), the DShot ranges for bidirectional/3D mode are:
- **REVERSE (backward)**: 48-1047
- **FORWARD**: 1048-2047

The value **0048 is the minimum reverse throttle value**, not a forward value.

**Question for @pitts-mo:**

When you transition from neutral to forward after being in reverse, what happens with the motor?

1. Does the motor **spin backwards** (reverse) briefly before going forward?
2. Or does it **spin forward** at low throttle as expected?

If the motor spins **forward** when receiving 0048, then yes, this is a bug (wrong value sent).

If the motor spins **backward** when receiving 0048, this might be intentional behavior - the code may be transitioning through the reverse range before entering forward range.

Could you clarify what you observe physically with the motor when this happens? A short video would be helpful if possible.

sensei-hacker, 2026-02-01:
I've done some research about it, but can't actually implement and test a fix until I'm done with another task.