iNavFlight/inav #10650: Can not complete gyro zero calibratinon if gyro sensityvity is higher then default.

and-sh, 2025-01-27:
## Current Behavior
Can not complete gyro zero calibratinon if gyro sensityvity is higher then default.

The is error that std dev is calculated from raw readings

[image]

and threshold is independent on gyro scale.

[image]
 

## Steps to Reproduce
Use precision gyro.

## Expected behavior
Zero calibration must be independent on gyro sensitivity.

## Suggested solution(s)
Adjust threshold on gyro scale factor.

## Additional context

All versions of Inav have this error

DzikuVx, 2025-02-03:
Which exact hardware this is related to?

and-sh, 2025-02-03:
It will be related to any hardware with resolution different from 16.4 LSB per degree/s.

Now I use custom board with icm45686. I found this problem when I was testing different FS for gyro and accel.

and-sh, 2025-02-04:
Second error of this type is accel clipping limit
https://github.com/iNavFlight/inav/blob/f00cd9ee37fa54f8674b17117bd88e34bf304e0b/src/main/sensors/acceleration.h#L30
When using an accelerometer with a limit of 32g or 8g, this will not work correctly.

If everything is left as is, these errors will cause a lot of difficulties in the future.