iNavFlight/inav #10659: Log corruption H7 F7 with rangefinder

rts18, 2025-01-29:
## Current Behavior
After the fix @bkleiner added for the H7 and F7 based boards. The card log now works correctly, even with `blackbox_rate_denom = 2`.
However I found a strange occurrence when I added a rangefinder to my quad. The log data becomes very broken up. Even when I lower the log denom rate to 32.

[image]

If I turn off the rangefinder serial connection. Logging again works at the higher rate with no corruption.

- FC Board name and vendor: tested with TBS_LUCID_H7 - MatekH743-WLITE  - MatekF765-WING
- INAV version string: 8.0.0

mmosca, 2025-01-29:
What range finder did you use? Was it MSP, or different kind?

rts18, 2025-01-29:
Its a Matek 3901-L0X module. 
It uses MSP because the module has lidar and optical flow.
I have tried disabling the MSP for the optical flow sensor. With the problem still remaining.

b14ckyy, 2025-04-11:
I can confirm. Matek 3901-LOX sensor and log is completely chopped. Already reduced log rate to 25% (500Hz) with same issue. Without rangefinder, a 50% Log rate was no problem on the same SD card.

INAV 8.0.1 Release. 

[image]