iNavFlight/inav #10660: Please, correct error in climb rate calculation.

and-sh, 2025-01-30:
## Current Behavior
Manual climb rate does not match the one specified in the configurator.

## Steps to Reproduce
Turn on althold. Measure climb rate. Check settings.

## Expected behavior
Correct climb rate.

## Suggested solution(s)
The problem is that on rcCommand deadband is applied twice.
First time in line 140
https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_multicopter.c#L140
second time in lines 149 and 153
https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_multicopter.c#L149
https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_multicopter.c#L153

You can fix it like this, for example.
[image]
It is tested and work well.

Removing deadband from lines 151 and 149 is not correct, since applyDeadbandRescaled changes ThrottleIdleValue in rcCommand. And it again produce a mess.

## Additional context
This problem was in all versions since 3.0.0. I didn't look further.

breadoven, 2025-01-31:
I think you might be right looking at it again. Probably explains why its sluggish to climb but drops suddenly when descending. This for hover throttle is < 1500. The opposite is probably true with hover throttle > 1500.

and-sh, 2025-01-31:
Well, yes, that's where I started to figure it out https://github.com/iNavFlight/inav/issues/10547. I've only just figured out how to fix it completely.

and-sh, 2025-01-31:
It seems that fixedwing also suffers from this error
https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_fixedwing.c#L112

breadoven, 2025-01-31:
Your solution above isn't right because it'll always be adjusting if there's no deadband applied, which you don't want. The issue is the way the adjustment is converted into a climb rate. Would need to check the numbers but it doesn't appear that max throttle gives you `max_manual_climb_rate `and min throttle -`max_manual_climb_rate `... you get something else.

Fixed wing is probably OK because it doesn't use a variable `altHoldThrottleRCZero `factor.

breadoven, 2025-01-31:
Actually checking this with debugs shows that the actual climb rate is just a bit higher than it should be but is consistent up and down. e.g. `max_manual_climb_rate `set to 700 actually results in +770 up and -770 down with `altHoldThrottleRCZero  ` set to 1320. So not exactly correct but not different up and down as I thought originally.

and-sh, 2025-01-31:
Yes, it is. But I don't understand anything again. Why is the tgtVel correct in this case?
The deaband was applied somewhere else. Need more time to check

[image]

[image]

and-sh, 2025-02-01:
But. This is incorrect too, because velocity starts not from zero.
May be this will be correct:
`        const int16_t rcThrottleAdjustment = applyDeadbandRescaled(rcCommand[THROTTLE] - altHoldThrottleRCZero, rcControlsConfig()->alt_hold_deadband, -500+getThrottleIdleValue(), 500);`

and-sh, 2025-02-01:
Although even so, perhaps
`        const int16_t rcThrottleAdjustment = applyDeadbandRescaled(rcCommand[THROTTLE] - altHoldThrottleRCZero, rcControlsConfig()->alt_hold_deadband, getThrottleIdleValue()-altHoldThrottleRCZero, getMaxThrottle() - altHoldThrottleRCZero);
`

breadoven, 2025-02-01:
I have a fix for this. I'll do a PR when I have time.

and-sh, 2025-02-01:
Ok. I already fix this for myself. 
On my setup difference in descent vel was more then three times.
max_manual_climb_rate was about 150

[image]

and-sh, 2025-02-01:
Fix this too, please, because errors tend to accumulate, and after some time everything stops working.

Max value of rcAdjustment is 500, but not 500-deadband.

[image]

and-sh, 2025-02-07:
It works fine, but I changed code a little for clarity
`int16_t rcClimbRate = rcThrottleAdjustment * navConfig()->mc.max_manual_climb_rate / ABS(limitValue);`