iNavFlight/inav #10666: MSP_STATUS_EX is sending only 2 bytes of armingFlags

JimB40, 2025-02-02:
Tested with INAV version 8.0.0

## Current Behavior
MSP_STATUS_EX is sending only 2 bytes of armingFlags
https://github.com/iNavFlight/inav/blob/f00cd9ee37fa54f8674b17117bd88e34bf304e0b/src/main/fc/fc_msp.c#L453

## Expected behavior
armingFlags are 32 bit so 4 bytes should be sent.

## Suggested solution(s)
sbufWriteU32(dst, armingFlags);

stronnag, 2025-02-02:
The correct solution is to use the correct API
* `MSP_STATUS_EX` is obsolete (as it predates 32bit arming flags)
* `MSP2_INAV_STATUS` is the modern equivalent that sends 32bit arming flags

stronnag, 2025-02-02:
See the [wiki article](https://github.com/iNavFlight/inav/wiki/MSP-V2#dealing-with-multiple-msp-versions) for advice on handling multiple INAV  / FC versions / variants in your application.

closed 2025-02-02 by stronnag

JimB40, 2025-02-02:
> * `MSP2_INAV_STATUS` is the modern equivalent that sends 32bit arming flags

INAV 8.0.0 does nlt support MSP V2 over telemetry as BF
So the only solution is to use V1

stronnag, 2025-02-02:
> > * `MSP2_INAV_STATUS` is the modern equivalent that sends 32bit arming flags
> 
> INAV 8.0.0 does not support MSP V2 over telemetry as BF So the only solution is to use V1

Then you will have to work with the limits of the INAV V1 API.