iNavFlight/inav #10667: Fix MSP V1 format - missing function byte in response

JimB40, 2025-02-02:
Tested on INAV 8.0.0

## Current Behavior

Current MSP V1 has bug and doesn't adhere to MSP V1 spec.
https://github.com/iNavFlight/inav/wiki/MSP-V2#legacy-and-compatibility

`function` byte is not send after `payload size`

[image]

This leads to numerous errors as we doesn't know if response coming is for actual request sent.
That leads to wrong data parsing.

## Expected behavior
`function` byte should be sent after `payload size` and before actual `payload` bytes

DzikuVx, 2025-02-03:
What exactly are we looking at? According to a MSPV1 frame outgoing from FC should begin with `$M>` that converts to HEX `24 4D 3E` and I don't see this sequence anywhere. Is this even MSP frame?

JimB40, 2025-02-03:
MSP V1 over telemetry. CRSF frame where '$M>' are stripped.
So first 3 bytes in this dump are CRSF specyfic.
4th is MSP packet length that
5th should be MSP function (missing)
Then actual MSP payload goes.

DzikuVx, 2025-02-03:
@JimB40 here is potential fix https://github.com/iNavFlight/inav/pull/10671 could you please test it?

closed 2026-10-04 by sensei-hacker