iNavFlight/inav #10681: MATEK H743 - INAV 8.0 - WRONG ALTITUDES

TheLoki2020, 2025-02-08:
## Current Behavior
When flying a newly built 8-inch drone with INAV 8.0 and tuned with EZ-TUNE, in the OSD the altitudes are completely wrong. From low altitudes when flying high to completely negative altitudes being reported. When trying to fly in POSHOLD the drone goes up or down in altitude without a real relationship when the throttle stick is in the middle.

[image]

All the logs of the last day of flying:
[LOG00097.TXT](https://github.com/user-attachments/files/18717844/LOG00097.TXT)
[LOG00098.TXT](https://github.com/user-attachments/files/18717848/LOG00098.TXT)
[LOG00099.TXT](https://github.com/user-attachments/files/18717847/LOG00099.TXT)
[LOG00100.TXT](https://github.com/user-attachments/files/18717845/LOG00100.TXT)
[LOG00101.TXT](https://github.com/user-attachments/files/18717851/LOG00101.TXT)
[LOG00102.TXT](https://github.com/user-attachments/files/18717852/LOG00102.TXT)
[LOG00103.TXT](https://github.com/user-attachments/files/18717850/LOG00103.TXT)
[LOG00104.TXT](https://github.com/user-attachments/files/18717846/LOG00104.TXT)
[LOG00105.TXT](https://github.com/user-attachments/files/18717849/LOG00105.TXT)

## Steps to Reproduce
1. Flash INAV 8.0
2. Tune the drone with EZ-TUNE
3. Fly in different modes (ACRO, POSHOLD, RTL)

## Expected behavior
Correct altitude values in the OSD and good POSHOLD altitude performance. 

## Suggested solution(s)
I do not know if the sensor fusion needs to be modified to solve this issue.

## Additional context
- DIFF: [https://pastebin.com/2318nDVB](https://pastebin.com/2318nDVB)
- FC Board name and vendor: MATEK H743 - SLIM V3 and Walksnail WS-M181 GPS/COMPASS
- INAV version string: ec2106af

breadoven, 2025-02-08:
Log 104 shows a complete mismatch between the Baro and GPS altitudes which likely explains the altitude problems. Hard to say which sensor is wrong, I'm guessing the GPS because it doesn't really increase even though climb is demanded and the Baro is increasing as you might expect. Try changing `inav_default_alt_sensor` to `BARO_ONLY `or `GPS_ONLY` to see it that improves things.

TheLoki2020, 2025-02-08:
Hi @breadoven next week I will try to fly with `inav_default_alt_sensor = BARO_ONLY` as the barometer information in the logs seems to be good. However, I tried to input the command in the CLI but I got the error of Invalid Name. What am I doing wrong?

[image]

sensei-hacker, 2025-02-08:
Another way of approaching this is to check a few things that will cause inaccurate altitude:

Missing foam will cause the baro to give inaccurate readings

A VTX or telemetry antenna mounted right on top of the GPS receiver will cause the GPS to lose accuracy

Things moving (wires that aren't zip tied, masts, loose screws) will cause the accelerometer to report inaccurate Z-axis movements.

So we can check for any wires that can flop around because aren't secured, ensure all screws and nuts are snug, make sure the baro foam hasn't gone missing, and ensure the GPS antenna is neither blocked nor has another antenna broadcasting into it. These checks SHOULD help ensure that all sensors provide the best possible data.

wyrwus, 2025-02-24:
> ## Current Behavior
> When flying a newly built 8-inch drone with INAV 8.0 and tuned with EZ-TUNE, in the OSD the altitudes are completely wrong. From low altitudes when flying high to completely negative altitudes being reported. When trying to fly in POSHOLD the drone goes up or down in altitude without a real relationship when the throttle stick is in the middle.
> 
> [image]
> 
> All the logs of the last day of flying: [LOG00097.TXT](https://github.com/user-attachments/files/18717844/LOG00097.TXT) [LOG00098.TXT](https://github.com/user-attachments/files/18717848/LOG00098.TXT) [LOG00099.TXT](https://github.com/user-attachments/files/18717847/LOG00099.TXT) [LOG00100.TXT](https://github.com/user-attachments/files/18717845/LOG00100.TXT) [LOG00101.TXT](https://github.com/user-attachments/files/18717851/LOG00101.TXT) [LOG00102.TXT](https://github.com/user-attachments/files/18717852/LOG00102.TXT) [LOG00103.TXT](https://github.com/user-attachments/files/18717850/LOG00103.TXT) [LOG00104.TXT](https://github.com/user-attachments/files/18717846/LOG00104.TXT) [LOG00105.TXT](https://github.com/user-attachments/files/18717849/LOG00105.TXT)
> 
> ## Steps to Reproduce
> 1. Flash INAV 8.0
> 2. Tune the drone with EZ-TUNE
> 3. Fly in different modes (ACRO, POSHOLD, RTL)
> 
> ## Expected behavior
> Correct altitude values in the OSD and good POSHOLD altitude performance.
> 
> ## Suggested solution(s)
> I do not know if the sensor fusion needs to be modified to solve this issue as the barometer data in the logs seems to be fine.
> 
> ## Additional context
> * DIFF: https://pastebin.com/2318nDVB
> * FC Board name and vendor: MATEK H743 - SLIM V3 and Walksnail WS-M181 GPS/COMPASS
> * INAV version string: [ec2106a](https://github.com/iNavFlight/inav/commit/ec2106af46720453f46855b23e4781f07fe79a93)

I had to disarm my drone when it flew over 300m up during the RTH procedure. Unfortunately, it crashed. In AltHold mode, it also behaved very unstably. It worked much better before with just GPS and the compass. I'm using iNav 8 stable. The drone is 10'.

sensei-hacker, 2025-02-24:
> I had to disarm my drone when it flew over 300m up during the RTH procedure. Unfortunately, it crashed. In AltHold mode, it also behaved very unstably. It worked much better before with just GPS and [...]

Did you check any of the things listed?
Wild baro fluctuations is strongly suggests your baro foam fell off.

TheLoki2020, 2025-02-24:
In the discord server, other users noted that I had a lot of accelerometer vibrations. I tried correcting them but I did not find a solution. In the end, I switched to betaflight. I only wanted the RTH function of Inav as I like it more than the betaflight but it is ok for me.

I will close this issue as for now I will not persist in finding a solution in Inav.

closed 2025-02-24 by TheLoki2020

tipoman9, 2025-03-09:
This happens very often to my wings also, unfortunately.

[image]