iNavFlight/inav #10683: Brushed esc problem

davifelix, 2025-02-09:
Good morning. I'm trying to get a bidirectional esc to work using a Kakute f4, inav 7 in rover mode, but I can't start it. Gps ok, compass ok, but I always have an alarm regarding the position of the throttle stick. I can start it from the output tab but I can't do it from radio. Has anyone ever tried and solved this problem?

sensei-hacker, 2025-02-09:
Please attach your diff.

It works fine for me, using default settings. Every few months someone reports they are having a problem with whatever settings they are using, but so far everyone has refused to provide the settings they are using, their diff. So I don't have any way of knowing which setting is causing their problem.

davifelix, 2025-02-09:
Thank you very much tomorrow I'll send you my diff....... Good evening

davifelix, 2025-02-10:
Entering CLI Mode, type 'ex
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# INAV/KAKUTEF4V2 7.1.2 May 31 2024 / 23:08:25 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 1 MOTORS
timer_output_mode 4 SERVOS

# Outputs [servo]

# safehome

# Fixed Wing Approach

# features
feature -TX_PROF_SEL
feature -AIRMODE
feature BAT_PROF_AUTOSWITCH
feature GPS
feature REVERSIBLE_MOTORS
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports
serial 0 0 115200 115200 0 115200
serial 2 0 115200 115200 0 115200
serial 3 2 115200 115200 0 115200
serial 5 64 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 900 1600
aux 1 45 2 1700 1900
aux 2 53 2 1900 2100
aux 3 10 1 1650 2100
aux 4 28 2 1075 1300
aux 5 31 2 1500 1700
aux 6 13 3 1675 2100
aux 7 30 2 1325 1475

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements

# master
set gyro_main_lpf_hz = 10
set gyro_zero_x = 2
set gyro_zero_y = 6
set gyro_zero_z = -9
set ins_gravity_cmss =  986.161
set acc_hardware = MPU6500
set acczero_x = 19
set acczero_y = -11
set acczero_z = 14
set accgain_x = 4086
set accgain_y = 4085
set accgain_z = 4033
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 182
set magzero_y = -157
set magzero_z = 121
set maggain_x = 450
set maggain_y = 462
set maggain_z = 491
set baro_hardware = BMP280
set min_check = 1550
set serialrx_provider = CRSF
set motor_pwm_protocol = STANDARD
set failsafe_procedure = DROP
set servo_pwm_rate = 100
set applied_defaults = 1
set gps_sbas_mode = AUTO
set gps_min_sats = 5
set nav_wp_max_safe_distance = 500
set nav_fw_loiter_radius = 100
set nav_fw_yaw_deadband = 5

# mixer_profile
mixer_profile 1

set platform_type = ROVER
set model_preview_type = 31

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 3 0 -100 0 -1

# profile
profile 2

# battery_profile
battery_profile 1

set throttle_idle =  0.000

# save configuration
save

# end the command batch
batch end

davifelix, 2025-02-11:
> Allega il tuo diff.

Hi, diff attached!

davifelix, 2025-02-12:
Hello everyone. Would anyone kindly help me? Thank you very much

sensei-hacker, 2025-02-13:
You can fix your current problem by entering in the CLI:
set min_check = 1100

You get arming blocked because you currently have:
set min_check = 1550

With max_check of 1900, you are essentially telling INAV that your radio is able to send values from 1550 - 1900.
It's looking for the center of that range, so 1725

If you set min_check back to the default of 1100, then it will see 1500 as center.

davifelix, 2025-02-13:
Thank you very much. I'll try, even if I think 1100 was the defoult value and it didn't work anyway. 
The esc I'm using to work needs the throttle stick being turned on to be set to about 1500. I can decide whether to have forward/reverse or forward/brake/reverse gear. Which setting might work? Forward/backward I think without brake function...........

sensei-hacker, 2025-02-13:
It is entirely possible that you will run into a DIFFERENT problem after you fix min_check.

If so, please describe the problem as precisely as you can and let me know what setting changes you made, turn I can see what's causing *that* problem.

All default settings pretty much works for me. The only changes I made for testing were setting bidirectional both in INAV and in the ESC.

At one point while experimenting with settings I ran into a combination of settings that could have a different problem. But anyway for now, let's see what you get after min_check is fixed.

davifelix, 2025-02-13:
HI. I set min check to 1100. now I have the problem that it gives me "unable to arm throttle is not low" how can it be low if I need it to be in the center to manage "forward/rear"? what do you recommend me to do??
Thank you

stronnag, 2025-02-13:
It is also the case that a number of  the Arming Disabled messages shown in the CLI and Configurator are bogus. See  #10635 and https://github.com/iNavFlight/inav/pull/10628

sensei-hacker, 2025-02-13:
> HI. I set min check to 1100. now I have the problem that it gives me "unable to arm throttle is not low" how can it be low if I need it to be in the center to manage "forward/rear"? what do you recommend me to do?? Thank you

What value do you see for throttle in the Receiver tab?
Around 1500 works for me, in 8 and I believe also in 7.1.2.

sensei-hacker, 2025-02-13:
> It is also the case that a number of the Arming Disabled messages shown in the CLI and Configurator are bogus. 

I suppose one is a number. :)
Yes, if you set up geozones, the message wasn't quite right. This diff does not have geozones.

sensei-hacker, 2025-02-13:
> The problem in my opinion arises from the fact that inav does not make me arm with 50% or 1500 pwm gas. In my opinion, if I solve this, I will have no problem arming. How do I set 1500 as default to arm?

Again, what value do you see on for throttle on the Receiver tab, please?

davifelix, 2025-02-13:
Unfortunately I didn't understand your question. in the receiver tab I see the value at which I leave the throttle stick. if I keep it at 1500 I see 1500.........

davifelix, 2025-02-14:
More tests done today but nothing. Unfortunately, he doesn't want to know about arming. I'm stuck........ 😥

sensei-hacker, 2025-02-16:
Actually the default min_check for 7.1.2 is **1000**
It's 1000 on INAV 8

I tested and confirmed, as you reported min_check =1100 will result in "throttle now low" on 7.1.2
For 7.1.2, you need the default:

set min_check = 1000
save

davifelix, 2025-02-16:
I did a test this morning and I managed to get the gear forward. I set the radio to have 0-100% stick 1500-2000 output signal e con uno switch posso avere sullo 0-100% 1500-1000 di segnale in uscita. On the cli I set min-check = 1500 e min commando = 1500 e finalmente ho ottenuto la marcia avanti. Non riesco comunque ad avere la retromarcia........

davifelix, 2025-02-16:
I'll do a test set to 1000 on min_check

davifelix, 2025-02-18:
Test if min_check Negative. I was only able to get the forward gear by setting min_check 1500 and min_commando 1500 below this value does not take any signal and I do not have reverse gear

sensei-hacker, 2025-02-18:
Arming works for me using the settings you have here
https://github.com/iNavFlight/inav/issues/10683#issuecomment-2649164305

Except with min_check put back to the default of 1000

Does arming work for you with those settings?

If you do two or three things at once, changing other settings to try to get other things to do what you want, I can't really help without knowing what you're doing at each step, and if THAT step worked.
When you jump two or three steps ahead and change multiple values I can no longer follow along and help you.

davifelix, 2025-02-18:
Ok, Could you tell me your settings as min_check and min_command? Type of esc (normal-brushed etc) and whether you have set the esc forward-brake-reverse or forward-reverse? Inserisco le tue stesse impostazioni e ti faccio sapere. Grazie mille per l'aiuto

sensei-hacker, 2025-03-14:
> Ok, Could you tell me your settings as min_check and min_command? 

I copy-pasted your settings from here, then set min_check back to the default of 1000
https://github.com/iNavFlight/inav/issues/10683#issuecomment-2649164305

That solves the problem of "unable to arm throttle is not low" - regardless of ESC.

If you put that back to the defaults and then you have a DIFFERENT problem with your ESC, we can then start solving your ESC problem.

To solve your ESC problem, we would first go to the Outputs tab, enable motors, and look at the value sent to the the ESC when the throttle is centered. You may want to either disconnect the motor from the ESC or disconnect the ESC from the FC for this check, to avoid having the wheel spin during the examination.

davifelix, 2025-03-14:
Thank you very much. I try and keep you updated!

davifelix, 2025-03-25:
Hello. I hope to be able to make myself understood well. Setting the min_check to the default value the Arm problem is solved. The real problem is this: when I turn on the vehicle, the fc expects the. Stick position to be at minimum to be able to arm. The esc however, to be able to center itself and activate would like the stick at 1500. This esc like all bidirectional escs, has reverse gear from 1500-1000 where 1500 is stopped and 1000 is at maximum reverse speed. the forward gear 1500-2000 where 1500 is stopped and 2000 is at maximum speed. So, with the settings like this, if I arm I will have the engine at maximum speed in reverse and this is not good...... How do I solve this problem?

davifelix, 2025-03-25:
Ok, How much do I have to set the min_command for?

sensei-hacker, 2025-03-25:
It works for me leaving it at the default, 1000.

davifelix, 2025-03-25:
👍👍👍👍👍Thank you very much for all the help. As soon as I've tried I'll let you know!