iNavFlight/inav #10685: Small error in gyro.c

and-sh, 2025-02-10:
## Current Behavior
Secondary dynamic notch executed even if dynamic notch disabled 

## Steps to Reproduce
Disable dynamic notch

## Expected behavior
Dynamic notch disabled

## Suggested solution(s)

[image]

## Additional context
Will affect performance a little.

and-sh, 2025-02-10:
A misprint can also be corrected

[image]

DzikuVx, 2025-02-11:
@and-sh the #10687 is the correct fix. I see you know C, you can fix this kind of things by yourself you know. It's open source after all

and-sh, 2025-02-11:
I can try. There is another issue https://github.com/iNavFlight/inav/issues/10674 here, seemingly obvious. I haven't figured out how it works yet.
What should I do?

PR #10687 (2025-02-18): Enable second dynamic gyro only if primary is enabled
closes #10685
closed 2025-02-18 by DzikuVx