iNavFlight/inav #10686: 8.0 auto disarm

fatbaldmen, 2025-02-10:
****

## Current Behavior
Inav 8.0 stable auto disarm  only works 2 times out of 10 after landing.

## Steps to Reproduce

1. Multiple Landings in one session with the same aircraft
2. Double checking throttle end points are 1000us on the low end and 2000 on the high end in the receiver tab of the configurator.
3. Waiting longer than the default a lot of time before Auto disarming gauges
4.

## Expected behavior

Auto disarm should engage after landing in the aircraft has been still for, I believe, the default 10 seconds regardless of what mode the aircraft is in

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

jaroszmm, 2025-02-10:
Same here. Sometimes it takes a while to disarm, there are occasions when it won't disarm at all if not using a switch. Can't find any particular pattern to it.

breadoven, 2025-02-11:
It probably means the pitch and roll angles aren't stable after landing. Check them in the OSD. If they're constantly changing even though the plane is static then it's probably related to horizon drift unfortunately. Also check the vertical and horizontal velocities, they need to be 0 or insignificant after landing, All of these factors are used to detect landing and auto disarm.

fatbaldmen, 2025-02-11:
> It probably means the pitch and roll angles aren't stable after landing. Check them in the OSD. If they're constantly changing even though the plane is static then it's probably related to horizon [...]

Thank you for the feedback. I checked my latest video, I do have pitch enabled in the OSD and after the aircraft is still pitch value only fluctuates 0.2°. But I do not have a roll value in my OSD, so I could be seeing drift there I will enable Ahi and the role of value and have another flight. I do have a motor that is causing resonance in the airframe, so Horizon drift is plausible. I will report back after I check those two things