iNavFlight/inav #10688: Weird indications of the compass

ViorelX, 2025-02-11:
I have installed Inav 8.0 on a Volantex Phoenix 2400 plane, FC Speedybee F405 Wing Mini, GPS unit MicoAir M10 ultra with QMC5883L compass.
I set the right alignment of the FC and of the compass, I calibrated the accelerometer and compass.
The heading indication was quite good initially, but often it changes radically up to 180˚.

How I proceed to fine adjust the compass:
On the flying field I direct the plane towards a reference object, in this case the church of a nearby village: 

[image]

I connect the telemetry viewer on the tablet:

[image]

The orange line is the indicated heading and the red arrow line, pointed to that church is the actual heading.
Then I connect to the Inav Configurator on laptop in order to correct the magnetometer alignment:

[image]

I think the difference between true and indicated heading is about 12˚, thus I change the yaw from the current 62˚ to 50˚. Then I press 'save and reboot'. It doesn't update immediately, so I repeated this task several times.
Finally, I got the updated yaw: 

[image]

But the newly indicated heading is totally wrong:

[image]

These are the magnetometer plots as I rotated the plane a complete circle around yaw axis:

[image]

After that I was flying and checking the navigations modes. It flies well, maintains straight path while cruising, loiters on almost perfect circles. Nothing to complain. But the heading indications are still totally wrong:

[image]

sensei-hacker, 2025-02-11:
What all is mounted within about 5 inches / 13cm of the magnetometer? Large power wires and buzzer will mess with the magnetometer.

ViorelX, 2025-02-11:
Good evening!
No, my problem has nothing to do with magnetic interference. The GNSS unit is far away from power wires. There are only some twisted wires for servos, passing 3-4 cm away. A wire carrying 1A generates 50uT of magnetic field (as strong as the earth magnetic field) at a distance of 4mm. My wires are 10x farther, carry less than 0.5A and are twisted.

[image]

Yes, the 62° or 50° are unlikely values for the compass alignment.
Initially I set the compass alignment to CW180flip and it was right. Then I went to the flying field and did some flight tests. It was fine. Only small corrections were required.
Next time I was flying, I noticed the compass indication was inverted!
Then I did several tests indoors, changing the compass alignment from CW180flip to CW0flip many times. Sometimes the compass indicates the right direction on CW180flip, other times on CW0flip, and it is often changing slowly. After a lot of adjusting, the alignment was left to that unusual 62°.
What I have presented is that a small correction of 12° leads to a huge modification of the indicated heading and there is not an immediate update, I have to insist a lot of times.
It is totally weird!!!

ViorelX, 2025-02-11:
I will check that empirical method in the next weekend.

kasatka60, 2025-02-14:
After turning the compass, do you recalibrate?

kasatka60, 2025-02-14:
In the inav configurator, does it turn when the plane tilts?

ViorelX, 2025-02-14:
My problem has been solved: following that empirical method, I have found the right GNSS unit orientation is CW0.
Now I have a rock stable compass!
It's so simple and it is ridiculous how much time I have wasted switching the orientation from CW180flip to CW0flip and vice versa.

Yes, after the last orientation setting I calibrated the compass again.
Thank you very much.

closed 2025-03-14 by sensei-hacker