iNavFlight/inav #10701: Switching VTOL flight mode not work (HW T-1)

Mauritz17, 2025-02-16:
I started with the masic masket set on the VTOL Page and copay paste it in the CLI.
I mixed all my controls right, and but all min max units in the output tab and get it mixed my engine Tilt mounts are in both Mixer profiles perfectly alligent. And moving corrispondely right.
I set up all Transition mode and Max mode for MC profile 2, and set up Max Mode for FW Mode.
2. I was set up 3 Pos switch and set the bars in the Modes Tab. (Set it on right aux chanel)
3. I was not calibrating the whole aero plane manually, i just paste the gyro Calibraiton files of prevus Projekt of this FC into the cli, now the "not calibrated warning" is gone.
I yoused an different ESC for the tail motor.
I flashed all ESCs ro the right Blue jay version, 16.7 or something. The Tail ESC have the Target Name "...H40" the front Motors "...H30"
I switched the output tab to Dshot125 all Motors spin ap same way on 11%
Will Inav verbinden the mode becurse mixed ESCs?
Should i make the Gyro Calibrating Process manual?
Both Micer Profiles are flight read setted up, all servos corisponding the ride way of RC input und by shaking the airplane. I have to click manual on "Mixer Profile 1/2" after this the FC reboots a d the Servos Tilt in right Position.
If i click on Inav PC on "Mix Profile 2" (my case Multicoper)  i can switch on middle Postion of Transition Swich, and the Servos going into Transition position. It doesent If I flick it in full Up/Down Position the Tilt goes Back to MC Mode.
I seen an setup video where someone went to output Tab hand take the Switch in this case there was an change of the Mdell shematic picture without rebooting. How i can get it done?

Jetrell, 2025-02-16:
>I seen an setup video where someone went to output Tab hand take the Switch in this case there was an change of the Mdell shematic picture without rebooting. How i can get it done?

That was likely done with video editing.. A flight controller reboot has to occur when changing between Control or Mixer profiles.

The best way to test if your mixer profiles are working. Is to remove the propellers then power the model on the bench to see if the motors and servo's react correctly, while not connected to the configurator.

>Will Inav decline the mode becurse mixed ESCs?

You need to run the same drive protocol on all your ESC's.

>Should i make the Gyro Calibrating Process manual?

Yes. You should always do a complete calibration of the IMU, for that specific flight controller.

Mauritz17, 2025-02-18:
The problem is solved. I plugged in my FC To the computer on the next day to try it again. And it worked by itselve. I  have no Idea why not on the privius day.. I dont know why not worked on the privosly day.