iNavFlight/inav #10702: Inav 8 angle mode drift

Tibor1111, 2025-02-16:
## Current Behavior

 
## Steps to Reproduce

1.
4 inch drone
weight with 3s lion bat. 240g
Flywoo GOKU GN F745 EVO HD
I know this problem is known. I want to point out that angle mode drift in Inav 8 is significantly worse than in Inav 7.1.2.
If I make a long turn in angle mode and then level the flight (sticks are in neutral) the drone remains tilted. It drifts. The drone levels the horizon in about 5-10 seconds. I did not have this problem in Inav 7.1.2.
I am attaching files

2.
3.
4.

## Expected behavior

[image]
[image]
[INAV_8.0.0_cli_20250216_115527.txt](https://github.com/user-attachments/files/18814695/INAV_8.0.0_cli_20250216_115527.txt)

What should I do to improve this?

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Tibor1111, 2025-02-16:
[MOV250203-163023-000632F - Trim.zip](https://github.com/user-attachments/files/18814729/MOV250203-163023-000632F.-.Trim.zip)
I am also attaching a video of the flight.

b14ckyy, 2025-03-03:
Please add your information to the oiginal report.

closed 2025-03-03 by b14ckyy