iNavFlight/inav #10719: too much files

sergey-prokhorov, 2025-02-22:
## Current Behavior
v.8
[image]

## Desired Behavior
v.7
[image]

## Suggested Solution
pack or something

## Who does this impact? Who is this for?
for all users

## Additional context
too much files!!!!

sensei-hacker, 2025-02-22:
How many of those files are in your build folder?

sergey-prokhorov, 2025-02-23:
> How many of those files are in your build folder?
there is no folder named "build"

sensei-hacker, 2025-02-23:
I see there's also no file called fr.pak.
You mean to be in the Configurator GitHub.

closure-util has a lot of files. Over half of them in .deps

sergey-prokhorov, 2025-02-23:
sorry, I meant configurator release. just downloaded in zip archive from GitHub.