iNavFlight/inav #10723: Visual Throttle Indicator in OSD

Carticiak, 2025-02-25:
## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

MrD-RC, 2025-02-25:
It would really be helpful to put a link to that video here. It's a bit tricky to understand what you're looking for by the description.

Carticiak, 2025-02-25:
> [@Carticiak](https://github.com/Carticiak) It would really be helpful to put a link to that video here. It's a bit tricky to understand what you're looking for by the description.

Sorry about that, I was trying to upload the video here, but it kept failing. here is the link, if that doesnt work I can upload it to youtube. 

https://www.facebook.com/share/v/1B59j4hsYt/

MrD-RC, 2025-02-25:
@Carticiak YouTube would be better. As some devs don't use Facebook. I'll see if I can check it out though.

MrD-RC, 2025-02-25:
Ah, ok the throttle makes sense seeing that.

What about the speed target? There is already a speed indicator for the graphical vario sidebars.  I guess you just want a target speed added. Rather than just showing if the speed is increasing or decreasing.

sensei-hacker, 2025-02-25:
I'm curious what the use case is for the throttle gauge?
In my own flying, I tend to know when I increase or decrease throttle.

MrD-RC, 2025-02-25:
@sensei-hacker it's not so much about seeing the throttle increase or decrease. It's more about at a quicker glance seeing how high up the sidebar the throttle indicator is. So halfway up the sidebar would be around 50% throttle.

Carticiak, 2025-02-25:
> I'm curious what the use case is for the throttle gauge? In my own flying, I tend to know when I increase or decrease throttle. Understanding the use case may be helpful.
> 
> A general gauge indicator, prefixed by a symbol to indicate what it is, could be interesting.

I find that more often than not, I'm incorrect in feeling where my throttle position is based on the sticks. I like this idea, maybe one or two gauge indicators that can be assigned to the user's preferred input. Similar to the battery indicator getting progressively more empty then turning red, that's really helpful to me when flying.

MrD-RC, 2025-02-25:
> > Ah, ok the throttle makes sense seeing that.
> > 
> > What about the speed target? There is already a speed indicator for the graphical vario sidebars. I guess you just want a target speed added. Rather than just showing if the speed is increasing or decreasing.
> 
> Basically like a speedometer in a vehicle, yes. Late 80's GMC style digital gauges come to mind. 

To be honest. I don't see that as useful. Those gauges always looked pretty tacky to me. Plus they had the numerical speed next to it. Which gives much better reference. I feel like with speed. Having a target speed and showing if you're above or below it may be a better indicator. But with speed I think knowing the actual number is better than on an arbitrary position on a scale.

Carticiak, 2025-02-25:
> To be honest. I don't see that as useful. Those gauges always looked pretty tacky to me. Plus they had the numerical speed next to it. Which gives much better reference. I feel like with speed. [...]

I have to agree with you, I don't know if I'd use a speed gauge like that, I just wanted to offer up more use cases for the same gauge. Maybe more of a novelty for a speedometer. 

At the very least, I'm hoping the devs could expand the custom OSD element spaces from 8 to maybe 16. I like my little gauge, but I'd like to set up something like that stall speed indicator as well.

MrD-RC, 2025-02-26:
The problem with stall is that it is not speed related. So indicators like that can be wrong. Stall is based on the airspeed over the wing and the angle the air hits the wing. Really for a stall indicator, you need a dedicated sensor. As they have on full size aircraft.