iNavFlight/inav #10729: GPS Failsafe

muhammedoney, 2025-03-03:
Hello, 

I have an FPV setup which includes a GPS on it. I was using it with GPS supported modes, and suddenly it fell down to the ground. I couldn't define the problem with neither logs or OSD, because it says disarmed by switch.

I think that it suddenly not able to take GPS sensor readings. Because before taking off, inav doesn't allow any GPS guided modes. I wonder if the GPS wasn't able to operate after the takeoff, will it trigger failsafe or simply falls down? 

And after fell off, i try to use my stack : even i switch of the GPS sensor from inav configurator, there was a ARMING_DISABLED_NAVIGATION_UNSAFE error. It seems like stack was stuck.

mmosca, 2025-03-03:
Disarmed by switch means that whatever channel you had used for the arm signal had changed value. 

What radio system are you using and how does it behave when your radio connection fails? If you configured your radio to set a fixed value for the arm channel it could easily cause you aircraft to disarm and drop from the sky.

You also have to remove any navigation modes for the modes tab, before removing the gps sensor.

Do you have dvr / blackbox logs of the flight?

muhammedoney, 2025-03-03:
Hello @mmosca 

I was using TBS Crossfire TX modul with my Jumper T15. And here is my log file, please check legend 14. On time 02:20 arm mode changes

[LOG66_LEGEND14_14.TXT](https://github.com/user-attachments/files/19052071/LOG66_LEGEND14_14.TXT)

mmosca, 2025-03-03:
Blackbox log has many flights, no idea which one you are reporting.

The last one I checked, disarmed at end of flight, with good rx status.

[image]