iNavFlight/inav #10742: INAV 8.0.1-RC1 DJI_NATIVE Wrong Icons

HishamGhosheh, 2025-03-07:
If I understood the [release notes](https://github.com/iNavFlight/inav/releases/tag/8.0.1-RC1) correctly then INAV 8.0.1-RC1 / Goggles N3 / DJI Air Unit O4 should no longer have problems with fonts

Updated N3 Goggles
[image]

Updated Air Unit
[image]

**Old behavior** 
Missing icons when using old BFHDCOMPAT OSD
[image]

**New behavior**
Wrong icons when using DJI_NATIVE OSD
[image]

**Expected behavior**
Correct icons and no question marks when using DJI_NATIVE 

FC version 
Compiled locally from https://github.com/iNavFlight/inav/pull/10070
`INAV/JHEF405PRO 8.0.1 Mar  7 2025 / 21:25:38 (GITDIR-N)`
`GCC-13.2.1 20231009`

Built in using WSL on Windows 11
[inav_8.0.1_JHEF405PRO.zip](https://github.com/user-attachments/files/19134859/inav_8.0.1_JHEF405PRO.zip)

mmosca, 2025-03-08:
Dji did not implement firmware auto detection. You need to configure the firmware type on the goggles as well. 

What you see is dji trying to use the betaflight font with inav.

HishamGhosheh, 2025-03-08:
Thanks! never knew about this option in the goggles: **Settings -> Display -> Flight Controller -> INAV**

It works perfectly now. Flying with question marks was not enjoyable and I almost switched back to Betaflight
[image]

closed 2025-03-08 by HishamGhosheh