iNavFlight/inav #10744: Add QMC5883P support

MATEKSYS, 2025-03-09:
QMC5883L is discontinued.
The substitute is QMC5883P,  https://www.qstcorp.com/en_comp_prod/QMC5883P.

https://github.com/ArduPilot/ardupilot/pull/25230

synersignart, 2025-08-02:
Is this still active for QMC5883P ?
can anyone pls check this commit and test

[compass_qmc5883p.zip](https://github.com/user-attachments/files/21559135/compass_qmc5883p.zip)

BortEngineerDude, 2025-08-02:
> Is this still active for QMC5883P ? can anyone pls check this commit and test
> 
> [compass_qmc5883p.zip](https://github.com/user-attachments/files/21559135/compass_qmc5883p.zip)

Submitting just these two files outside of version control won't cut it. Take a look at [this PR](https://github.com/iNavFlight/inav/pull/10994)

closed 2025-08-03 by MATEKSYS