iNavFlight/inav #10745: Loitering banking while rudder output is oppsite?

majianjia, 2025-03-10:
[image]
The plane is banking right while the rudder is turning left. 

Hi,

I just want to discuss yaw control during a banking turn.

I flew last weekend, and while shooting videos, I noticed that my V-tail plane looked strange in the air—kind of struggling while loitering in RTH (Return to Home) mode.

I’m used to flying wings (which have no rudder), so I don’t naturally use the rudder most of the time. The plane was under power during the flight. I am still tuning the plane so it could be some issue with my setting, too. 

I’m wondering if I might have set the mixer in the opposite direction or if INAV uses a different turning logic for a good reason.

Thanks!

sensei-hacker, 2025-03-10:
Opposite rudder is often needed. Commonly, you use rudder in the direction of the turn to *initiate* the turn, then a touch of opposite rudder *during* the turn to keep the nose level with the horizon and coordinate the turn.

The amount of differential aileron you set will only be correct for one airspeed and one rate of turn, so any other speed or rate of turn will require rudder in one direction or the other, to avoid either skidding or slipping (having the aircraft flying sideways through the air).

To ensure you have it set the right direction, first look in the Receiver tab. Moving *only* the yaw axis, ensure the correct channel (the yaw channel) goes up when you push yaw right.

On the set-up tab, ensure the 3D model moves the same way you move the actual aircraft. Check each of the three axis independently.

Then in manual mode, move the yaw stick and ensure the control surfaces move in the correct direction. If not, invert the weight for the yaw line in the mixer.

Lastly, in Angle mode, wiggle the aircraft in yaw and see that the control surfaces make small movements in the correct direction.

> I don’t naturally use the rudder most of the time

It's a useful thing to learn, a good habit. Coordinated turns are one of the first things you learn and something you practice a lot as a full-size pilot. If a pilot isn't using it he rudder, they are skidding their turns, having the aircraft move one direction while pointed a different direction, so it's pushing sideways through the air. This increases stall speed, while also making the stall more dangerous. As well as increasing drag and wasting power.

majianjia, 2025-03-10:
@sensei-hacker 
Thanks for the info! It makes sense to me.
It looks like the plane was underpowered while loitering, which caused the opposite yaw control. During my video shooting, the rudder was mostly opposite to the aileron.
I believe I have set up the mixer correctly now.
I used to fly DLG, where I relied on the rudder a lot. Then I switched to FPV flying wings, where there was no rudder response at all.
Just havent got back to flying a regular plane

closed 2025-03-10 by majianjia