iNavFlight/inav #10749: Barometer failure in INAV 8

msu2313, 2025-03-11:
## Current Behavior
Barometer fails during flight, briefly spikes then remains at zero for the remainder of the flight. This issue began after upgrading from INAV 7.1.2 to 8, and is still present in 8.0.1.  See below pics that show the OSD the moment of barometer spike and then the zero out reading for the rest of the flight. These pictures are 1 second from each other in time. 

This is on a Mini Drak, my GPS is roughly 15 inches away from my DJI O4 air unit. All the wires inside the fuselage are tied off and not loose. 

[image]

[image]

## Steps to Reproduce

1. Fly in Acro mode 
2. Steep dives and tight turns
3.
4.

## Expected behavior
I expect that the barometer will continue to register altitude and climb rates (As it did in INAV 7.1.2) 

## Suggested solution(s)

## Additional context

See attached CLI and Black Box log

[INAV_8.0.1_cli_MINI_DRAK_20250310_203211.txt](https://github.com/user-attachments/files/19175035/INAV_8.0.1_cli_MINI_DRAK_20250310_203211.txt)

[LOG00036.TXT](https://github.com/user-attachments/files/19175055/LOG00036.TXT)

---
# version
# INAV/MATEKH743 8.0.1 Feb 25 2025 / 10:29:03 (171d00bb) 
# GCC-13.2.1 20231009

- FC Board name and vendor:  Matek H743 WLITE
- INAV version string:

Oleggif, 2025-03-12:
Подтверждаю, у меня такая же проблема. Контроллер speedybee F7V3.

SimonFPV01, 2025-05-12:
Using SpeedyBee 405 Wing with DJI VTX 04 Pro and DJI Goggle 3 after upgrading from iNav 7.1.2to iNav 8.0 and 8.0.1 the Altitude and Home direction and distance did not work. After changing the Speedybe 405 Wing and GPS, still it did not help, including updating DJI google 3 and DJI VTX 04 to latest firmware. I downgraded iNav back to 7.1.2 (leaving the DJI 04 and Goggle with the latest) this fixed the problem.