iNavFlight/inav #10750: RC Throttle doesn't go to motor idle, but goes to zero when disarm

funnybrum, 2025-03-11:
## Current Behavior
When disarmed servos that get their input by the `RC Throttle` goes to 1000us.

## Steps to Reproduce
1. Configure servo mix for a servo to be RC Throttle
2. Arm and observe how servo moves correctly with RC throttle.
3. Disarm and servo goes to 1000us.

## Expected behavior
RC Throttle should go to the configured value for `Reversible Motors Neutral`.

## Suggested solution(s)
Use `Reversible Motors Neutral` instead of the 1000us when disarmed.

## Additional context
I need precise RC Throttle output directly to an PWM controlled ESC. Without all the filtering that the motor output does. Using the RC Throttle seems to be one way for this, but when I disarm the rover goes in full reverse.

- FC Board name and vendor: SpeedyBee F405 v4
- INAV version string: INAV/SPEEDYBEEF405V4 8.0.0 Jan 21 2025 / 16:16:11 (ec2106af)

sensei-hacker, 2025-03-11:
Yes, that's a known issue. It had been addressed, then not.
Maybe we will address it again.

For now, a workaround you can use is to set a global variable to the throttle value (in the Programming tab), then use the global variable in the servo mixer.

funnybrum, 2025-03-11:
Thanks @sensei-hacker, the proposed solution did the job.

Leaving a screenshot from my programming tab. May not be the optimal approach, but at least it seems to work fine.

[image]

This is no longer issue for me. I'll let the maintainers judge if this is a bug that deserves fixing or not.

funnybrum, 2025-03-11:
The approached worked after a few more experiments. Global variable to servo PWM mapping is not 1:1. To control servo PWM with global variable you need the global variable to be in the -500 to 500 range. GVAR value of -500 maps to 1000us PWM and GVAR value of +500 maps to 2000us PWM. The math required to get this working was: (RC throttle - 1500).

Or as follows:
[image]