iNavFlight/inav #10752: Please, consider a tiny improvement to compass_rm3100.c

and-sh, 2025-03-12:
1. A check new reading is unnecessary waste of time because sampling rate of compass is much more than reading rate.

 So
[image]
before
[image]
after
[image]

2. Sampling rate can be substantially lower than default. PNY suggest at least 400 for minimum noise  
[image]
I set 550 now, that correspond to about 58Hz of sample rate. 
[image]
it give noise about 1 in 3 digit.

https://github.com/user-attachments/assets/b1e0230a-f3be-4106-a10e-800eae2db169

I guess that sampling rate can be lower, but it can lead to errors due to sequential sampling of magnetic field vector components.

and-sh, 2025-03-12:
QMC5883
need about the same time to read

[image]

The noise is about an order of magnitude higher

https://github.com/user-attachments/assets/b9296194-add9-4cca-ab7f-b46edf12d21e

b14ckyy, 2025-04-09:
Double Post? https://github.com/iNavFlight/inav/issues/10748

closed 2025-04-09 by b14ckyy

and-sh, 2025-04-09:
Nope, https://github.com/iNavFlight/inav/issues/10748 is about incorrect order of register initialization.
The https://github.com/iNavFlight/inav/issues/10752, is just a performance improvement.

b14ckyy, 2025-04-09:
Since it both affects the same hardware and your titles are very generic, please condense both requests in one post so we keep the GH a bit cleaner. 
I will do a complete sweep the next days and clean up a lot as well to prevent a mess and not lose track of important tickets.

and-sh, 2025-04-09:
done
https://github.com/iNavFlight/inav/issues/10812