iNavFlight/inav #10756: INAV 8 deadly AHI Drift when GPS signal is lost

tipoman9, 2025-03-13:
INAV/SPEEDYBEEF405WING 8.0.0 Jan 12 2025 / 19:18:44 (f0fcd155) 
I wrote about the problem [here](https://github.com/iNavFlight/inav/issues/10588),  but this time I can provide all possible logs.
CRSF Log [FPV-2025-03-09_AR_Wing_Pro_AHI_Drift.csv](https://github.com/user-attachments/files/19237618/FPV-2025-03-09_AR_Wing_Pro_AHI_Drift.csv)
INAV BlackBox [LOG00070.TXT](https://github.com/user-attachments/files/19237565/LOG00070.TXT)
INAV settings dump [INAV_8.0.0_cli_20250314_000314.txt](https://github.com/user-attachments/files/19237566/INAV_8.0.0_cli_20250314_000314.txt)
[Video as a file](https://drive.google.com/file/d/1WwYq9GOrjZ4SH95o2BOofLNFjSe59Xp-/view?usp=sharing) - to sync it with the blackbox data, enter an offset of -230 in blackbox explorer

There is no suspicion of GPS spoofing - the flight path does not deviate from the real one and "lands" exactly where I had taken off.
Here is an [online version of the attached video](https://youtu.be/OU11vFFL_1k).

Here is another case of [AHI drift  immediately after GPS](https://youtu.be/dDb1sNGmgcI?t=78) is lost:

I think this should be addressed - it means almost certain crash if the GPS is lost. I was able to land only because I was flying LoS.
And yes, the plane does not vibrate and the same airframe [does 30km flights smooth as silk](https://youtu.be/UIl2ZlZwFA8) when there is GPS coverage

Picture for attention - Notice the pitch and roll offset...
[image]

MrD-RC, 2025-03-14:
Please do not open a new issue for the same problem. This should have been appended to the previous issue.

MrD-RC, 2025-03-14:
Closed as duplicate

closed 2025-03-14 by MrD-RC

Jetrell, 2025-03-14:
@tipoman9 What GNSS module are you using ?
Your satellite signal quality is still all over the place, as it was in your other post.

But something else I picked up in the log and settings. Is how poor the magnetometer heading is.
The first thing to note about this is how bad your magnetometer calibration is.
If a magnetometer is available, the AHRS will use it to compensate for accelerometer drift.. So if the mag calibration is off, or it experiencing magnetic deviation. Then the AHRS will receive poor correction data.
And being that both GNSS/mag are poor. Have you considered that the plugs or cable between the GNSS module and the FC has an intermittent connection ?

tipoman9, 2025-03-14:
> Please do not open a new issue for the same problem. This should have been appended to the previous issue.

@MrD-RC the previous issue is about DeadReckoning and all cruise modes, as described in #10536
This seems to have been addressed and the issue can be recognized as solved.
Now I'm trying to explain a different problem, that I only partially mentioned there, so that's why I considered it deserves a new ticket.
Please reopen the issue, since it makes the use of DeadReckoning option that was added in INAV 8 useless and even dangerous.

tipoman9, 2025-03-14:
> But something else I picked up in the log and settings. Is how poor the magnetometer heading is.

The module is Beitian BE-252q. 
I spend a lot of time calibrating the magnetometer and I still consider it a plausible cause for the problem since I experience a similar problem with two different planes that both have magnetometer enabled (since I need the DeadReckoning functionality) .
The GPS+compass module is placed in the wing, near the tip, AR WingPro has a dedicated compartment for this. The 6-pin cable to connect it to the FC is rather long (about 70cm), but since magnetometer info goes as digital signal via I2C, I don't think data corruption is possible.

tipoman9, 2025-03-14:
@Jetrell can you give me some clues how to analyze discrepancies of the magnetometer data in the log? I see the metrics, but I'm not sure to what other metrics I should corelate them.
Thanks

MrD-RC, 2025-03-14:
> I wrote about the problem here, but this time I can provide all possible logs.

> Now I'm trying to explain a different problem, that I only partially mentioned there, so that's why I considered it deserves a new ticket.

You are contradicting yourself. If it's not about the problem in #10588 why mention it at all. Looking at Jetrell's comments. The issue seems to be related to the same causes. You are flying in a GPS denied area and have bad compass calibration. From what I see, it is the same issue.

stronnag, 2025-03-14:
> You are contradicting yourself. If it's not about the problem in [#10588](https://github.com/iNavFlight/inav/issues/10588) why mention it at all. Looking at Jetrell's comments. The issue seems to be [...]

Exactly. If you replay the log, you can see the plane jump sideways and backwards from the the GPS location. All the while with number of satellites and HDOP all over the place.

Adjacent to a war zone and in the vicinity of military bases ... hardly an INAV problem.

tipoman9, 2025-03-14:
> You are contradicting yourself. If it's not about the problem in [#10588](https://github.com/iNavFlight/inav/issues/10588) why mention it at all.
Because while troubleshooting the other issue I also provided data for the AHI drift. I clearly stated there:
_So, seems there are two issue I'm facing.
1. When navigation mode is engaged, pitch oscillations occur - this seems the be addressed here https://github.com/iNavFlight/inav/issues/10536
2. In case of GPS outage (or maybe spoofing), inertial navigation may get affected._

tipoman9, 2025-03-14:
@stronnag INAV has included the Dead Reckoning option, **which is supposed to work exactly when GPS outages occur.**
Now you and @MrD-RC claim that **since I experience GPS outages, this is not an INAV problem**?!
If this is not a contradiction, then what, catch 22? 
The war zone is 400km away, as you can see from the logs provided.
I calibrated the magnetometer several times according to the manuals - if someone can help me with the problem in its calibration, I will gladly make the changes and test them in flight.

sensei-hacker, 2025-03-14:
There are two types of things that could be causing the problems with your GPS and mag readings.
Either the problem is:
a) Onboard the aircraft (extra noisy VTX, etc.)
or
b) externally caused (such as a radar dish somewhat nearby)

 We can perhaps get much closer to a solution by cutting the possible causes in half. Which can be done by flying in a different place, as far as possible from where you encountered the problem. For example, one area where I tried flying, I has major problems that would suddenly appear and then go away, appear and then go away. It turned out, there was a radar antenna not too far away and when the dish was pointed toward my aircraft it really messed things up. Flying in a different area helps remove external causes from consideration.

As I say that, be aware that GPS interference can be somewhat widespread. Here's the current data:

[image]

> INAV has included the Dead Reckoning option, which is supposed to work exactly when GPS outages occur.

There may be a misconception or two (or three) here.  Dead reckoning attempts to keep flying when the GPS fails, rather than performing an emergency landing, in the hopes that the GPS comes back within a few seconds. It does that by relying more on the magnetometer and IMU, when the GPS stops providing data.

That doesn't mean it will work well when the GPS _and the mag_ are sending **wrong** information.
In fact, there are three different ways it would be wrong to say that it will work well in that situation:

1. It doesn't mean it will work well in any case - only that it will try, rather than doing an emergency landing.
2. The GPS sending wrong info is not the same as an outage, as the GPS either cutting off or reporting no fix.
3. Without GPS, wrong mag data will be extra problematic - it means most of the info being fed to INAV is wrong

If you do some flying on the other side of town, please let us know what you find there. If the problem doesn't appear on the other side of town, it's likely external interference of some kind. If the problem is the same no matter where the aircraft is, then the problem is likely onboard the aircraft.

PS:
> rather long (about 70cm), but since magnetometer info goes as digital signal via I2C, I don't think data corruption is possible.

Digital data can be and often is thoroughly corrupted. The difference between analog and digital in this respect can be seen if you compare analog video vs digital video.  You may have seen with analog you'll get little specks of "static" noise on the screen. With digital, the entire bottom half of the video may suddenly go away pretty much much completely.

 Analog has an infinite range of values, so it can get a little interference fairly easily.  A value that is supposed to be 1.892 can instead read 1.894. Analog data will be a little bit wrong when problems happen. Digital data has discreet steps, similar to integers that can only be 1, 2, 3, 4 etc.  When digital data gets corrupted/interference, a value that is supposed to be 4 will instead be 8 or 1 or 254. It most often won't be just a little bit wrong, it will be a **lot** wrong.

Digital doesn't mean the data will always be right. Digital means that when it works well, a value that's supposed to be 4 will be exactly 4 - not 4.002. And when it's wrong, when there is interference or corruption, that 4 will read as 200 or something, not as 3.9976

daijoubu, 2025-03-14:
All electrical signaling is analog.  When the signal is interpreted as binary (digital) it can be more immune to corruption, but with enough signal degradation it will get corrupted.  This image shows the difference, the signal on the left can be easily sampled with clean highs and lows and good transitions in between.  On the right, the highs are not always high and the rise/fall time is inconsistent and drawn out, so it's unclear when to sample the voltage to determine if it's a high or low bit.

[image]

tipoman9, 2025-03-14:
@daijoubu I thought the protocol over i2c implements some data integrity like a parity bits or CRC sum. Checked the QMC5883 and seems this is not the case, so this is a plausible cause, but I would speculate this will mean erratic data if this happens.

@sensei-hacker 
At the moment if you lose GPS, you can safely control your plane home. If you happen to also lose RC link (or are in navigation mode by chance), the plane will start emergency landing ([I almost lost a plane this way](https://youtu.be/s7Rdtn-8hV0)).
So DeadReckoning is a useful feature that should prevent the latter - and that's why I try to make use of it.
BUT, if the AHI drift problem persists, it turns into a drawback, **since when it is enabled, when you lose GPS you also get AHI drift, that will make you plane crash even if you have solid Video and RC link.**

I investigated the mag[Y] metrics and it seems rather inconsistent when the plane is flying in one direction without changing orientation much.
Assuming the magnetometer orientation is correctly setup in INAV (I used the dot on the QMC5883 IC as a reference ), what should Y axis represent? Pitch, Roll, Yaw.
Here is how the value changes for just a couple of seconds.

[image]

[image]

Jetrell, 2025-03-14:
>can you give me some clues how to analyze discrepancies of the magnetometer data in the log? I see the metrics, but I'm not sure to what other metrics I should corelate them.

I described it in this wiki [link](https://github.com/iNavFlight/inav/wiki/GPS-and-Compass-setup#compass-calibration-and-testing). What range the calibration _zero offset_ and _gain_ should be within.
Then compare them to yours, which are far from that.
`maggain_x = 1102`
`maggain_y = 783`
`maggain_z = 1237`

> BUT, if the AHI drift problem persists, it turns into a drawback, since when it is enabled, when you lose GPS you also get AHI drift, that will make you plane crash even if you have solid Video and RC link.

Please refresh yourself with what I wrote to you yesterday, about how the mag is used to correct accelerometer drift. Which it can't if the mag heading is incorrect.

>Assuming the magnetometer orientation is correctly setup in INAV (I used the dot on the QMC5883 IC as a reference ), what should Y axis represent? Pitch, Roll, Yaw.

I'd assume calibration or align_mag is not correct.  XY are not representative of a specific direction, due to the multiple ways a user can mount the module/mag in their model. The orientation preset can be corrected by way of the `align_mag` setting.

tipoman9, 2025-03-15:
@Jetrell 
Thank you for your time. I did read all the links and this subtle notice seems interesting.
***A sensor flip is always to realize over the pitch axis. For example cw270flip:***
So, my setup was
```
set align_mag_roll = 1800
set align_mag_pitch = 0
set align_mag_yaw = 0
```
Now I changed it to *do the flip over he pitch axis*
```
set align_mag_roll = 0
set align_mag_pitch = 1800
set align_mag_yaw = 1800
```
With both settings I get the same UI for orientation

[image]

And now doing the test with rotating the plane over the yaw and observing the change of the bearing seems to perform much better than before.
Can this change affect the magnetometer performance?

Jetrell, 2025-03-15:
>Can this change affect the magnetometer performance?

It certainly can If the magnetometer isn't working in unity with the IMU.. Have a good read through that wiki document. It will answer most of your questions.

We haven't seen your build. And you said the GNSS module is out on the wing. That is a long cable run.
Its always recommended to twist all the wire together on long cable runs, to help prevent the likelihood RFI or EMI from other onboard devices interfering with the serial and i2C bus.. And in your case, something is certainly messing up the GNSS data.