iNavFlight/inav #10763: Matek F405-WSE randomly lost configuration!

dzaro-dev, 2025-03-17:
## Current Behavior
INAV 8.0.0 randomly lost configuration on a Matek F405-WSE. The same happened some time ago on INAV 7 and the plane became unresponsive and dropped out of the sky. When connecting the FC to the PC and running the configurator, the new setup dialog box popped up and all previous configuration was lost.

## Steps to Reproduce
This has randomly happened several times, no particular steps to reproduce, just regular usage. 

## Expected behavior
FC should not loose configuration after everything is setup and saved.

## Suggested solution(s)
More testing required before a release.

## Additional context
https://pastebin.com/X3n5KMpU

daijoubu, 2025-03-17:
Are you unplugging the DC connector/USB connector while it is writing to flash such as when you disarm with the odometer enabled?

Sibre3, 2025-03-17:
>Suggested solution(s)
More testing required before a release.

@dzaro-dev That's a bit rude and initialed. You said in your own words _"This has randomly happened several times, no particular steps to reproduce, just regular usage"._

How do you expect developers to test every target after every feature submission!? That's the reason for release candidates.

mmosca, 2025-03-25:
Configuration is lost, usually when you unplug while it is writing changes to flash.

This happens when you disarm and have things like continuous auto trim or odometer enabled.