iNavFlight/inav #10764: No telemetry received from AM32 esc

pmarkiewicz, 2025-03-17:
Replaced BLHeli_32 with AM32 (latest fw)
Connected T from Esc to Rx on FC
Enabled one on ports page.
Set temperature and rpm on osd.

And nothing, it was working with blheli but not with am32.
AM32 claims to support "KISS standard ESC telemetry" no idea if one is different from protocol used by blheli

undiplomatic, 2025-07-23:
Have you enabled serial telemetry in the ESC firmware?

undiplomatic, 2025-07-23:
> Replaced BLHeli_32 with AM32 (latest fw) Connected T from Esc to Rx on FC Enabled one on ports page. Set temperature and rpm on osd.
>
> And nothing, it was working with blheli but not with am32. AM32 [...]

Have you enabled serial telemetry in the ESC firmware?

undiplomatic, 2025-07-25:
> there is no option to enable/disable telemetry.
> I only have selected 30ms interval telemetry.
> 
> regard
> […](#)

That should enable it. Mine is also not working. Not sure if I can be bothered to connect a scope to the UART wire.

undiplomatic, 2025-07-26:
> I don't have scope unfortunately
> […](#)

I've ordered a spare loom so I can splice into it, connect a scope and try to diagnose the fault.

pmarkiewicz, 2025-07-28:
ok, connected ftdi and there is transmission, I got data below, 115k 8N1
FE FE FF FE FF 
9F BF BF 7E FF FF BE DF BF FF DE BB BF FF FF DF 9D 7E CB DF 9F DF FF BF BF 3E BB BB DB B3 BA BB EF BF BB BF 7F F3 BB EF 9F 
9F BB FB 7F BB BB FB BF BF BB 6E 3B DF 3B BE FF FE FF 7F 3E FF BF FF 3F FB 7F FF BF 7B BF 7F FF DF F3 7E FE 7F DF F3 FB FB 7F FF AF BE 9D 
B3 3A FF 7F 9F 9E FF 1E 7F 1E FB B7 DD 9F 7E 7B FF FE DF FB FB BF FF 3E BD FB FB DF FF 1E FE 3B 3A BB BA FE FF DF DE BB ED BA BE 7A 7F FF 
FF FF 3F 7F 3A BB 3A DE BB 9F FF BF DF BB 7F 7B 3A BB DF 5E FB FF 3E BF FF 7E DF BF FF FF FB 7F FF BF FF DE BF BF FF 9E FF 9E DE FB BF DF 
FF FB FF 3E AF FF 9E DB BF 1E BF FF DF 7F BF BF FF DE 3E EF BF BF FF FE FF 9E FB BE FF 7F DF AF BE BD FF DE BF FF FF AF BF 7F BF DE 9F 7F 
DF 3E BF DF BF EF FF DF BF 9F BB 7F AF BF 3E F3 7F F7 FF 5E BD CD 3A FB 7E BF 9F 7E BB 3B 7B FF 5E BF BA BE 7B 7E FF DF BF DE FF BF DE DF 
DD FE FF DF 37 BF FF 9E EF FB FD 9F 9F 6F 7F BF F7 FF DE FF FF FF AF BB BD 76 FE B7 7E DF FF DF 5E BF AF 7F DF 3F FB 3F 7F FF DF FE DE BF 
2F 3A 3B 3E 3B BF FF DF F3 7B 7E BF BF BF F3 BB 3A BB 77 FF DF BF BB FF AF BB 7F 3A F3 3A FF DF BF FF 73 BB 3B 3A FB DF FF DF FF B7 3B FF 
FF DF 7F BF EF FF FF DF FF DF 3B FB