iNavFlight/inav #10765: OLED last line corrupted

ProfDrYoMan, 2025-03-18:
## Current Behavior
The OLED shows a corrupted lowest line.

[image]

## Steps to Reproduce
Connect OLED to I2C and enable in configurator.
Display as shown above.

## Expected behavior
It works? :)

## Suggested solution(s)
No idea.

## Additional context
Display type:
https://www.aliexpress.com/item/1005006501624032.html?spm=a2g0o.order_list.order_list_main.5.21ef1802E8UjmP
128X64 I2C SSD1306 12864

- FC Board name and vendor: Speedy Bee F405 wing mini
- INAV version string: INAV 8.0.0

ProfDrYoMan, 2025-03-18:
Oh, sorry.

b14ckyy, 2025-04-09:
Will be fixed with this. You can test the Artefact if you want: https://github.com/iNavFlight/inav/pull/10760

sensei-hacker, 2025-05-31:
I've learned there are a few different chips used for OLED drivers.

The SSH1106 needs more data sent to it than the SSD1306.  The SSD1306 just gets bytes for the pixels.  The SSH1106  has memory for four extra lines that are not displayed. I may do a PR to try to detect which kind of display is attached and adjust appropropriatrly.