iNavFlight/inav #10769: Need access to GPS Time of Day

NorthIceMan, 2025-03-19:
## Current Behavior
Current Accessing in iNAV: While the UTC time is used internally by iNAV (e.g., for logging purposes in blackbox or OSD display), it is not directly available as a selectable operand in the Programming Tab's logic conditions (e.g., no "UTC Hour" or "UTC Minute" or "UTC Seconds" variable).
 
## Desired Behavior
Get access to UTC (Time of Day) somewhere in iNAV.  Make it available in Programming and let the User do the rest.
Have "UTC Hour" , "UTC Minute" and "UTC Seconds" variable OR one variable (all in seconds) and let User manipulate it in Programming Logic.

## Suggested Solution
Make Time of Day (thru GPS)  available in iNAV Programming Tab.  How about in "Flight" Programming slot?  Or maybe in "Flight Mode" (one of the 4 USERx slots)? Or a new slot?

## Who does this impact? Who is this for?
All Users

## Additional context
Time of Day can be displayed on OSD but cannot be accessed in iNAV.  Needed to include in those great Programming logics.
Either the UTC time or the GPS local corrected time would be fine.
This would be very useful for timestamping anything User generated and could be sent in telemetry to update Time of Day in Tx at boot.
Benefit: This ensures your transmitter’s clock is always accurate and aligned with UTC, which could be useful for logging, timing flights, or coordinating with other systems.
Consequence of not having UTC accessible in iNAV - While some very expensive newer Tx's may provide access to UTC this does not allow people who don't have these newer Tx's to have access to UTC

sensei-hacker, 2025-03-19:
What is your actual use case, please?

NorthIceMan, 2025-03-19:
It started with the need to have the Time in my Tx on boot.  But then I see many other uses like 1. Being able to reach a WP at a certain time 2. Having it in OSD data when taking flight videos 3. measuring distance between WPs in minutes and seconds
The list goes on.  I do long range flights over forest and reaching the end point is often at a camp where the "greeting party" would know what time the FW will arrive

sensei-hacker, 2025-03-19:
> It started with the need to have the Time in my Tx on boot.

Then you would need it in the *telemetry*. Some telemetry protocols have that field, some don't. Having it in the programming tab doesn't redefine the telemetry protocol to include it, of course.

The OSD has multiple time display options, including GPS time.

NorthIceMan, 2025-03-19:
> that field, most don't. Having it in the programming tab doesn't redefine

I use FrSky Telemetry .  Its not just a telemetry need though - its very
interesting what a long range hobbyist can do with the correct time.  Like
stop flying (AutoLand) when local protocols don't want any models/drones in
the sky after a certain hour.  Can save getting ticketed.
Lets just say that if "the developers" want to do this for us well we
should get a CROWD to donate a little gold!

Ron

sensei-hacker, 2025-03-19:
Okay I assume that by "Frsky telemetry" you probably mean Smartport, not the classic Frsky protocol from long ago?

Smartport has fields for longitude, latitude, course, HDOP, and some others - but not time, from what I see. So no matter what we do in INAV, your Frsky radio won't accept the time over Smartport telemetry. Unless you intend to do some kind of hack?

NorthIceMan, 2025-03-19:
Yeah-I agree but don't discount the need for the UTC in Programming - I can handle the Telemetry after with scripts

sensei-hacker, 2025-03-26:
> Yeah-I agree but don't discount the need for the UTC in Programming - I can handle the Telemetry after with scripts

If you do that script let me know - I would be curious to see it.

closed 2025-05-13 by sensei-hacker