iNavFlight/inav #10770: Please, add GPS Heading to OSD, this great aid for compass tuning for everyone.

and-sh, 2025-03-21:
for example by this way

[image]

MrD-RC, 2025-03-21:
I'm pretty sure there is already a ground course OSD element.

breadoven, 2025-03-21:
There is ground course but it doesn't use the GPS value, instead it calculates it from estimated x, y velocities. Should be pretty much the same thing.

sensei-hacker, 2025-03-21:
> There is ground course but it doesn't use the GPS value, instead it calculates it from estimated x, y velocities. Should be pretty much the same thing.

That's what ground course is. There's no such thing as GPS heading. There's GPS course.

and-sh, 2025-03-21:
Oh. Does not not found it and start searching from gps. Of course this is the same values.

[image]

It is interesting that this value go to DJI osd instead of COG.

closed 2025-03-21 by and-sh