iNavFlight/inav #10773: Pitch oscillation with the default position setting in INAV 8

majianjia, 2025-03-23:
[image]

My plane was an 800mm V-tail model, but the tail was quite short. So I increased the pitch mixer to 70%. 
I'm not sure if that made it worse... 
Unfortunately, I didn’t get a chance to record video or take photos before it crashed.

After auto tuning, the pitch FF increased and the oscillation is also getting worse. 

BlackBox Log

sensei-hacker, 2025-03-23:
If / when you get it rebuilt, let us know how the glide test goes. A lot of things can cause pitch instability, especially without much tail volume to compensate.  Having the CG behind the center of pressure may be the most common problem. That will cause a feedback loop that turns the slightest upset into an ever-increasing pitch moment.

majianjia, 2025-03-23:
The CG is fine as I did manually fly this plane. The I saw the pitch oscillation. 
Github doesnt allows me to share the log maybe it was too big, 19M

Just check this long discussion. Seems related to the changes from alt position to rate. 
https://github.com/iNavFlight/inav/issues/10536#issuecomment-2566250218

Will rebuild the plane and fly next weekends probably. Will try to reduce the throw, alt P and response. 
# get nav_fw_alt_control_response 
nav_fw_alt_control_response = 40
Allowed range: 5 - 100
# get nav_fw_pos_z_p 
nav_fw_pos_z_p = 25
Allowed range: 0 - 255

breadoven, 2025-03-23:
It seems to be oscillating in pitch when Nav is idle according to that BB log screenshot so this can't be related to Althold issues if that's the case. Or is that BB log screenshot misleading, what mode was it in  ?

majianjia, 2025-03-29:
[image]

Just did a flight with the rebuilt plane. This time, I reduce the throw to default 50% at pitch. 
The plane fly well now in acro and manual mode, but not in angle or RTH.

For angle mode and RTH, it is bumpping quit heavily. Not a good weather today, quite windy.

majianjia, 2025-03-31:
As you can see in the log, the throttle also oscillating badly. fw_alt_out is clipped at maximum 200 during RTH. 

[image]

breadoven, 2025-03-31:
A Diff file of the settings would probably be most useful

majianjia, 2025-04-01:
@breadoven Please see below for the diff. 

```
$X>q 

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/SPEEDYBEEF405WING 8.0.0 Jan 21 2025 / 16:16:30 (ec2106af) 
# GCC-13.2.1 20231009

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]
servo 1 1000 2000 1557 100
servo 2 1000 2000 1557 100
servo 3 1000 2000 1476 100
servo 4 1000 2000 1524 100

# safehome

# Fixed Wing Approach

# geozone

# geozone vertices

# features
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y
blackbox SERVOS

# Receiver: Channel map

# Ports
serial 4 33554432 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 1500 2100
aux 1 1 1 900 1300
aux 2 12 1 1725 2100
aux 3 10 3 1725 2100
aux 4 21 4 1675 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 46 6 V
osd_layout 0 1 33 2 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 7 24 15 V
osd_layout 0 8 19 17 V
osd_layout 0 9 1 9 V
osd_layout 0 11 39 2 V
osd_layout 0 12 47 1 V
osd_layout 0 14 2 8 V
osd_layout 0 15 1 1 V
osd_layout 0 20 42 16 V
osd_layout 0 21 42 17 V
osd_layout 0 24 16 0 V
osd_layout 0 28 47 10 V
osd_layout 0 30 13 14 V
osd_layout 0 32 27 2 V
osd_layout 0 34 21 0 V
osd_layout 0 35 47 2 V
osd_layout 0 39 1 6 H
osd_layout 0 46 46 15 V
osd_layout 0 85 2 12 V
osd_layout 0 100 2 17 V
osd_layout 0 109 46 5 V

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = 4
set gyro_zero_y = -5
set gyro_zero_z = 7
set ins_gravity_cmss =  980.970
set gyro_adaptive_filter_min_hz = 20
set acc_hardware = ICM42605
set acczero_x = -3
set acczero_y = -1
set acczero_z = -9
set accgain_x = 4097
set accgain_y = 4097
set accgain_z = 4094
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set pitot_hardware = VIRTUAL
set blackbox_rate_denom = 8
set motor_pwm_protocol = STANDARD
set align_board_pitch = 18
set small_angle = 180
set applied_defaults = 3
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_ublox_use_beidou = ON
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 800
set nav_wp_max_safe_distance = 500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 40
set nav_fw_loiter_radius = 5500
set nav_fw_control_smoothness = 2
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_spinup_time = 250
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set nav_fw_allow_manual_thr_increase = ON
set osd_video_system = AVATAR
set name = BM POLY-TALON

# control_profile
control_profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set tpa_rate = 0
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 4

# mixer_profile
mixer_profile 1

set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 28

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 -100 0 -1
smix 1 2 0 -100 0 -1
smix 2 3 1 -50 0 -1
smix 3 3 2 -50 0 -1
smix 4 4 1 50 0 -1
smix 5 4 2 -50 0 -1
smix 6 1 10 -50 20 -1
smix 7 2 10 50 20 -1

# battery_profile
battery_profile 1

set throttle_idle =  5.000
set nav_fw_max_thr = 1800
set nav_fw_launch_thr = 1900
set nav_fw_launch_idle_thr = 1200

# save configuration
save

# end the command batch
batch end
```

majianjia, 2025-04-01:
According to the thread you guys discussed in the vertical rate PR, here are some highlights.
They were set by the configurator.

```
# TPA need to be set but it was disable.
set tpa_rate = 0

# increase smoothness should prevent overshoot, but i was at default 2
set nav_fw_control_smoothness = 2
```

Jetrell, 2025-04-02:
@majianjia After looking at the log. It appears that your `fw_ff_pitch` is over driving the rate setpoint. Which in turn is causing the vel_z target to oscillate.
It appears you have not ran an Auto Tune yet. Because the Feedforward and Rates are still default.

A Feedforward of 70 on the pitch is too high for your build. I'd take it down to 50 for a start, then reduce it by 10 after each test. It should be gone by a minimum of 30.. But the pitch stick response maybe a bit more sluggish if you go below that... It  will all depend on your elevator control surface throw. If its already low.

majianjia, 2025-05-05:
After a few more crash-rebuild cycle.. like every weekends from my first post. 

The plane was finally flying smooth in pitch and other axis now. 

> The BlackBox are always fail to parse with Inav 8.0.1 (most of the time) while it is ok in Inav 7. This doesnt help at all. it was better when I set it to higher data rate (1/2) now never record successfuly in 1/8. Not updating because none of the log was record successfully.  

It wasnt too much pitch FF or throw in the previous discussion. I tried to reduce FF to 50, then crashed twice, one just after launch, second crash was cant pull up just a few second after launch.  Just dont have enough pitch authority. 

Below is the final diff for my plane. 
TBO, the problem is the new rate control in 8.0 vs position control in 7.

key changes

```
set nav_fw_control_smoothness = 6
set nav_fw_alt_control_response = 25
```

``` 
diff all

# version
# INAV/SPEEDYBEEF405WING 8.0.0 Jan 21 2025 / 16:16:30 (ec2106af) 
# GCC-13.2.1 20231009

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Timer overrides

# Outputs [servo]

# safehome

# Fixed Wing Approach

# geozone

# geozone vertices

# features
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y
blackbox SERVOS

# Receiver: Channel map
[... 224 lines cut ...]
set throttle_idle =  5.000
set nav_fw_max_thr = 1800
set nav_fw_launch_thr = 1800
set nav_fw_launch_idle_thr = 1200

# battery_profile
battery_profile 2

set throttle_idle =  5.000

# battery_profile
battery_profile 3

set throttle_idle =  5.000

# restore original profile selection
control_profile 1
mixer_profile 1
battery_profile 1

# save configuration
save

# 
```

majianjia, 2025-05-13:
[bonafidepirate](https://www.youtube.com/@bonafidepirate) 
also mentioned in his video  https://youtu.be/rcfCzrWs2n4&t=449 that in INAV 8, pitch got oscillation while INAV7 is fine. 
I observed the same oscilating frequency as in his video. 
It isnt cause by overtune pitch FF or P/D in my testing (in fact, autotune advices more pitch FF in my setup).

Jetrell, 2025-05-14:
> It isnt cause by overtune pitch FF or P/D in my testing (in fact, autotune advices more pitch FF in my setup).

Did you note in my last post, how I said - **"It will all depend on your elevator control surface throw".** 
It can be caused by either, but it's generally a slower oscillation if the FF gain is too high. 

Autotune should derive the required tune, from the maximum rate of rotation your plane is capable of. 
This is going to be determined by the amount of control surface throw you have set; the airspeed you conduct the auto tune at. And also how well the [Autotune](https://github.com/iNavFlight/inav/wiki/Modes#autotune-fw) was conducted. 

The vel_z controller feeds through the main PID controller and then the mixer, to drive the servo's, and perform its task.. This is why `fw_ff_pitch` has to be tuned correctly.

majianjia, 2025-06-19:
Hi @Jetrell 

`stick_input * (FF + throw + control surface) < expected pitch rate`  So autotune add more FF, which is expected. 

After I checked the changes log from INAV7 to 8 and a few PRs. I realized that the pitch oscillation i observed is altitude oscillation, not because the pitch PID was overtuned. 
Then I reduced `nav_fw_alt_control_response` in my last flight and the problem now solved. 
 
No blackbox was sucessfuly recorded, so these statements are from my observation. 

Thank you guys. I like the rate control altitude control idea but the default value can be improved. 
I will close this issue now but looking forward for improvement too.

closed 2025-06-19 by majianjia