iNavFlight/inav #10790: Walksnail headtracker no data

BortEngineerDude, 2025-03-29:
## Current Behavior
Hello, I am having an issue with the Serial Headtracker. Walksnail system is unable to provide headtracking data over MSP. Mixing Headtracker tilt to the servo output yields nothing. Debug graph with debug_mode = HEADTRACKER shows nothing but zeros. 

## Steps to Reproduce
1. Configure Walksnail Googles X to PTZ headtracking
2. Configure VTX serial port to MSP Displayport
3. Set Headtracker to MSP
4. Mix Headtracker tilt to a servo output
5. Set dubug_mode to HEADTRACKER

## Expected behavior
1. Servo gimbal is tracking head movements
2. Debug graph shows headtracker data

## Additional context
I have:

Walksnail Googles X flashed with firmware v38.43.13 (latest, as of now), headtracking is set to PTZ

Walksnail Moonlight VTX connected via MSP Displayport, flashed with firmware v15.1.17 (latest, as of now)

A simple single-servo tilt gimbal mixed to a spare PWM output. Adjusting the output value manually or mixing the output with any RC input makes the servo move, as expected.

As a sanity check, I have switched headtracker on the googles to the PPM mode and hooked it up to a remote, using trainer mode to verify, that googles are, in fact, capable of headtracking. On the drone itself I have attached a scope to a FC receive pin of a serial port dedicated to VTX, and managed to observe some data going back into FC from VTX.

Am I missing something here? I am at a loss.

---

- FC Board name and vendor: SpeedyBee F403 V4
- INAV version string:

mmosca, 2025-03-29:
Is your osd working and does recording start when armed?

BortEngineerDude, 2025-03-29:
Yes, I can see INAV HUD in the googles and it does start recording on arming, on both sides.

mmosca, 2025-03-29:
Have you updated the headtracker firmware on the goggles?

BortEngineerDude, 2025-03-29:
Oh, wait a minute, do googles have a separate firmware for a headtracker? I am supposing that it has built-in headtracking capability, and I don't have any external modules.

mmosca, 2025-03-29:
Yes, they do. Make sure you running at least 3.4.

BortEngineerDude, 2025-03-29:
Seems like it will take some time. Do I need a special adapter to upgrade the built-in headtracker? Or maybe, do you know the pinout of the upgrade cable? (Assuming it is using UART, where's GND, RX and TX pins?)

BortEngineerDude, 2025-03-29:
Yep, outdated headtracker firmware was the issue. It had firmware v1.7, upgrading it to v1.9 made it work, thanks for your help.

In case of anyone else is having the same issue, here's how can you solve it: follow this guide - https://www.youtube.com/watch?v=hP3qzn777qk

Tip: having a special upgrade adapter is not necessary, you can hook up the AV cable from the Googles X kit to a generic 3.3v USB-UART adapter like that:
[image]
[image]

closed 2025-03-29 by BortEngineerDude

mmosca, 2025-03-29:
Next time you order something from caddx, add this to your order: https://caddxfpv.com/collections/all-cable/products/gl-and-gm-upgrade-cable  Should make life easier. ;)