iNavFlight/inav #10795: TARGET  NEUTRONRCF435WING  Uart5 BUG

sunyanmeng963, 2025-04-01:
When I was using NEUTRONRCF435WING, I encountered a bug related to the serial port. It uses the AT32 microcontroller. In the target file, it defines four serial devices namely Uart3, Uart5, Uart7 and USB. However, in the ground station, only Uart3, Uart5 and Uart7 are displayed. It seems there is a problem. I carefully reviewed the code but couldn't find out where the problem lies. Could you help me figure out where the problem is?

sunyanmeng963, 2025-04-01:
In the AT32F435/437 series, there are four built-in universal synchronous/ asynchronous transceivers (USART1, USART2, USART3 and USART6), and four universal asynchronous transceivers (UART4, UART5, UART7 and UART8). Could this be the cause of the problem?

sunyanmeng963, 2025-04-01:
The problem has been solved. There was no total erasure when downloading.

b14ckyy, 2025-04-09:
Closed as resolved

closed 2025-04-09 by b14ckyy