iNavFlight/inav #10798: STM32F405RGT6  connect terminal by 115200, CLI mode not responding

tigrajack, 2025-04-02:
1) I executed the command:
"make MATEKF405"

2) and flash inav_8.0.1_MATEKF405.hex into STM32F405RGT6

3) connect terminal by 115200

4) type "#"

and and I get nothing in return

sensei-hacker, 2025-04-02:
What happens if you use the CLI tab in Configurator and type "status"?

tigrajack, 2025-04-02:
> What happens if you use the CLI tab in Configurator and type "status"?

disconected after 5-6 seconds
I think, what in src/main/target/MATEKF405/target.h 
#define USE_UART1
#define UART1_RX_PIN            PA10
#define UART1_TX_PIN            PA9

#define USE_UART2
#define UART2_RX_PIN            PA3
#define UART2_TX_PIN            PA2
is not true pins

sensei-hacker, 2025-04-02:
If the pins were wrong, nobody with a Matek F405 would have been able to use those UARTs for the last eight years. Given that thousands people have been using those boards successfully for three last eight years, we can be pretty confident the pins are correct.

I take it you have connected a serial adapter to uart 1 and set MSP on that uart?

tigrajack, 2025-04-02:
> If the pins were wrong, nobody with a Matek F405 would have been able to use those UARTs for the last eight years. Given that thousands people have been using those boards successfully for three last eight years, we can be pretty confident the pins are correct.
> 
> I take it you have connected a serial adapter to uart 1 and set MSP on that uart?l, or
> 
> Can we see how you wired that up, please, and a screenshot of your Ports tab?

No i use stm32f405rgt6 (not Matek F405) 
I use CDC option.
For betaflight it is work.
Please tell me the default uart pin for stm32f405rgt6.

sensei-hacker, 2025-04-02:
The Matek F405 is one of many, many flight controllers with a stm32f405rgt6 chip. Any flight controller with "405" in the name is using that chip.

I'm guessing what you mean is you are trying to make your own flight controller?

USB CDC doesn't have anything to do with UART1 or UART2.

tigrajack, 2025-04-02:
> The Matek F405 is one of many, many flight controllers with a stm32f405rgt6 chip. Any flight controller with "405" in the name is using that chip.
>
I already test many flight controller with stm32f405rgt6 - without successfully.
 
> I'm guessing what you mean is you are trying to make your own flight controller?
No i am already made own  flight controller. I am try integrate it with your inav configurator.

> 
> USB CDC doesn't have anything to do with UART1 or UART2.
platformio.ini
       build_flags =
		-D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
		-D USBCON
it is work for UART0 default for Serial
> 
> If you are trying to make your own flight controller, but were not aware that F405 is the _chip_ and aren't clear on what a UART is - designing your own flight controller might be too advanced of a project for your current understanding.

I am not very good at expressing my thoughts in English, but I am good at understanding hardware. 
To be precise, I use a board WeAct STM32F4 based on chip stm32f405rgt6 ( https://github.com/WeActStudio/WeActStudio.STM32F4_64Pin_CoreBoard )

I have a very simple question for you: How to find out which pins are used in stm32f405rgt6 by default for WeArt board.
If you can help, I will be grateful.